raspberrypi/documentation #1155: FAQ prevent accidental corruption

ita1024, 2019-06-02:
While the on/off switch section describes that [SD card corruption can happen](https://github.com/raspberrypi/documentation/blob/master/faqs/README.md#where-is-the-onoff-switch), there are no guidelines on how to make a safe setup (`do not turn it off` is unrealistic).

A section called `How do I prevent accidental corruption?` would solve this, and it could mention the following:
* Reducing the amount of card writes by disabling the swap (`dphys-swapfile swapoff; dphys-swapfile uninstall; systemctl disable dphys-swapfile`), enabling logging in RAM (journald.conf) or disabling systemd-timesyncd
* Making Raspbian read-only, ideally with a link to a recommended overlay setup like [this](http://wiki.psuter.ch/doku.php?id=solve_raspbian_sd_card_corruption_issues_with_read-only_mounted_root_partition) or [this](https://k3a.me/how-to-make-raspberrypi-truly-read-only-reliable-and-trouble-free/)
* Using a UPS with a watchdog

JamesH65, 2019-09-18:
Added some text in this PR.

https://github.com/raspberrypi/documentation/pull/1285

ita1024, 2019-09-19:
Looks great to me! :tada: 

Do not hesitate to close this issue once the changes are merged.

JamesH65, 2019-11-06:
Merged.

closed 2019-11-06 by JamesH65