raspberrypi/documentation #1158: Raspberry Pi Camera Module - advice to use rpi-update needs caution adding? https://www.raspberrypi.org/documentation/raspbian/applications/camera.md

DeeJay, 2019-06-07:
> "If you've checked all the above issues and the Camera Module is still not working, you may need to upgrade the firmware on the Raspberry Pi. Use the following command to get the very latest (but experimental) firmware:
> 
> sudo rpi-update"

Perhaps needs a 'health warning' that this experimental firmware might render the installed OS unstable? Discussion in this forum thread suggests that rpi-update is deprecated in almost all circumstances - [https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=241977](url)

JamesH65, 2019-06-07:
I've just taken a look at this, I'm not sure. It does say experimental, and it's a last resort IIF you camera isn't working.

JamesH65, 2019-06-07:
OK, I decided just to remove that and subsequent paragraph.

closed 2019-06-07 by JamesH65