raspberrypi/documentation #1182: configuration/hdmi-config.md: Pi4 specific sentence doesn't make sense

andrum99, 2019-06-25:
The second paragraph of this page says:

> When using the Pi 4, which has two micro HDMI ports, you can attach two 1080p60 displays, or two 4Kp30 display (to HDMI port 0)

You can only connect one display per port, so the bit in brackets doesn't make sense.

lurch, 2019-06-25:
Looking at the [history](https://github.com/raspberrypi/documentation/commit/f5d7f09c04aa9422fb72ae754219e3cb5d82394d) of the page, you can see where the error snuck in. Looks like "two 4Kp30 display (to HDMI port 0)" should be "two 4Kp30 displays" ?

andrum99, 2019-06-27:
> Looking at the [history](https://github.com/raspberrypi/documentation/commit/f5d7f09c04aa9422fb72ae754219e3cb5d82394d) of the page, you can see where the error snuck in. Looks like "two 4Kp30 display (to HDMI port 0)" should be "two 4Kp30 displays" ?

That's a reasonable guess, but I would rather someone who knows what the sentence should say makes the decision, since I don't.

JamesH65, 2019-06-27:
Removed the errant brackets and also clarified it can be one or two devices. The way it read sort of implied you had to have two!

JamesH65, 2019-06-27:
Please close if you are happy with the change.

lurch, 2019-06-27:
I guess the text _could_ still be misread that the Pi4 doesn't support resolutions below 1080p60? :man_shrugging:

andrum99, 2019-06-27:
> I guess the text _could_ still be misread that the Pi4 doesn't support resolutions below 1080p60? 🤷‍♂

It could, but I will close this issue and address it in a PR.

closed 2019-06-27 by andrum99

andrum99, 2019-06-27:
And that PR is #1187.