raspberrypi/documentation #1183: Error with hostapd after following the instructions

dazz100, 2019-06-26:
Hi
I followed the instructions but I have followed the instructions at 
[https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=235145&p=1485851#p1485851](url)
down to, but not including "Add routing and masquerade".

This included update and upgrade to:
Raspbian GNU/Linux 9 (stretch)
Kernel: Linux 4.14.70-v7+
hostapd v2.6
running on a Raspi 3B+

I get the following error message:

sudo systemctl status hostapd.service
● hostapd.service - Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator
   Loaded: loaded (/lib/systemd/system/hostapd.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Wed 2019-06-26 21:11:52 NZST; 1s ago
  Process: 28548 ExecStart=/usr/sbin/hostapd -P /run/hostapd.pid -B $DAEMON_OPTS ${DAEMON_CONF}(code=exited, status=1/

Jun 26 21:11:52 ap-name systemd[1]: hostapd.service: Control process exited, code=exited status=1
Jun 26 21:11:52 ap-name systemd[1]: Failed to start Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authen
Jun 26 21:11:52 ap-name systemd[1]: hostapd.service: Unit entered failed state.
Jun 26 21:11:52 ap-name systemd[1]: hostapd.service: Failed with result 'exit-code'.
lines 1-9/9 (END)
● hostapd.service - Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator
   Loaded: loaded (/lib/systemd/system/hostapd.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Wed 2019-06-26 21:11:52 NZST; 1s ago
  Process: 28548 ExecStart=/usr/sbin/hostapd -P /run/hostapd.pid -B $DAEMON_OPTS ${DAEMON_CONF} (code=exited, status=1/FAILURE)

Jun 26 21:11:52 ap-name systemd[1]: hostapd.service: Control process exited, code=exited status=1
Jun 26 21:11:52 ap-name systemd[1]: Failed to start Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator.
Jun 26 21:11:52 ap-name systemd[1]: hostapd.service: Unit entered failed state.
Jun 26 21:11:52 ap-name systemd[1]: hostapd.service: Failed with result 'exit-code'.

This post is to provide evidence of a problem (the instructions don't end up with a working AP).  The root cause may be an issue/bug with hostapd.

[https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=235145&p=1485851#p1485851](url)

Other have had the same problem reported here:
[https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=235145&p=1485851#p1485851](url)

Others have fixed this by downgrading to version hostapd 2: 2.4-1 + deb9u2
I haven't tried this yet.  That is the next step.

dazz100, 2019-07-01:
Rather than upgrading from Stretch Lite, I have tried configuring an AP from a fresh installation of Raspbian Buster Lite 2019-06-20.
I followed the same instructions.  The AP appeared to work.  I could see it from other clients using the command:  
`sudo iwlist wlan0 scan`
and I could see the SSID from my phone. This is progress.  
I have not yet been able to log in to the Raspi AP.  Still working on that.

Dazz

lurch, 2019-07-19:
IIRC the docs are only targeting the latest released Raspbian (i.e. Buster at the time of writing).

@JamesH65 Should this be closed?

JamesH65, 2019-07-19:
I think so - the instruction work for the current release and we don't maintain docs support for older releases. 

Glad it's working!

closed 2019-07-19 by JamesH65