raspberrypi/documentation #1185: Pi 4 B doesn't give correct address/size for peripherals as per documentation

picosonic, 2019-06-26:
In the documentation at [https://www.raspberrypi.org/documentation/hardware/raspberrypi/peripheral_addresses.md](https://www.raspberrypi.org/documentation/hardware/raspberrypi/peripheral_addresses.md) it suggests using **bcm_host_get_peripheral_address()** and **bcm_host_get_peripheral_size()** to get the peripheral address and size to ensure future compatibility.

However on the Raspberry Pi 4 B, these return the following incorrect values :

```
peripheral address : 0x0
peripheral size : 0xfe000000

```

When these values are then passed to **mmap()** it fails.

Could this be related to the change of format for the `/proc/device-tree/soc/ranges` file.

JamesH65, 2019-06-26:
Almost certainly, the get_peri address function checks in /proc/device-tree/soc/ranges for address and size.

@pelwell How has the DT changed here?

pelwell, 2019-06-26:
The physical address has grown to be 64-bits wide, even though the top 32-bits will be zero, as I think @picosonic already understands. The userland library needs updating accordingly.

JamesH65, 2019-06-26:
Is it as simple as updating the final four functions in https://github.com/raspberrypi/userland/blob/master/host_applications/linux/libs/bcm_host/bcm_host.c to use 64bit values?

Or, to retain backwards compatibility, should we add new functions specifically for 64bit?

lurch, 2019-06-26:
Probably worth copying the way that https://github.com/RPi-Distro/raspi-gpio/commit/80fa7d04eafb3ea34fc6f2d32de5f1873b5fb369 does it?

pelwell, 2019-06-26:
The "correct" way to parse the DT is to read the properties "#address-cells" and "#size-cells", then calculate offsets and sizes using those values. However, a reasonable approximation in this case is for both bcm_host_get_peripheral_address and bcm_host_get_peripheral_size to read offset 4 first and add 4 to their usual offset if the result is 0. Hence bcm_host_get_peripheral_size always does two reads - offset 4, then offset 8 or 12 - while bcm_host_get_peripheral_address only needs to read from offset 8 if the read from 4 is zero.

tangentsoft, 2019-08-14:
I just want to leave my vote that libraspberrypi be updated for the Pi 4: we shifted to use this library instead of direct twiddling of `/proc/device-tree/soc/ranges` as in the raspi-gpio code linked above, and now we find that the version of the library shipped with Buster for the Pi 4 doesn't return a value you can correctly use on the Pi 4.

We'd prefer to continue using generic Raspberry Pi Foundation provided interfaces rather than do things the way the raspi-gpio project does it.

While in there, could you provide something like the get_gpio_pull() and is_2711 flag in the raspi-gpio project's solution to this problem? (Again, RPi-distro/raspi-gpio@80fa7d04e)

Thanks!

lurch, 2019-08-18:
This has been confirmed as fixed - see https://github.com/raspberrypi/firmware/issues/1161 for details.

closed 2019-08-18 by lurch