raspberrypi/documentation #1191: Is PIP Included in the (Minimal) Lite Raspbian Image?

radilly, 2019-06-29:
This regards https://github.com/raspberrypi/documentation/blob/master/linux/software/python.md

Helpful page.  I've not used pip yet, preferring apt-get if at all possible as suggested (and have been able thus far to use only "stock" Python).   All my Pi's are running the Stretch Lite version, save one which is Jessie Desktop, and it looks to me as though the desktop version may be the only one which includes pip?  The page hasn't been updated recently and only mentions Wheezy and Jessie.  I'd guess the pattern continues with Stretch (and maybe Buster).  Could one just say that the Lite versions do not include pip?

Cheers, and Thanks,
Bob

JamesH65, 2019-07-01:
@bennuttall Any thoughts?

bennuttall, 2019-07-01:
The problem is, pip requires `build-essential` which brings in a ton of dependencies. The lite image intends to be as minimal as possible, and pip is non-essential. @XECDesign 

I will update the python guide - thanks for the heads up :+1:

closed 2019-07-01 by bennuttall

radilly, 2019-07-01:
Thanks Guys ... Hadn't thought about large dependencies.  That's a great reason not to include it!

;-)

lurch, 2019-07-01:
@bennuttall @spl237 Does that page also need updating to now recommend using Thonny, now that the Buster images don't have IDLE pre-installed?

bennuttall, 2019-07-01:
This one does: https://github.com/raspberrypi/documentation/blob/master/usage/python/README.md

lurch, 2019-07-01:
@bennuttall Oops yes, that's the page I meant. In my haste I didn't realise that @radilly was talking about `linux/software/python.md` (which doesn't mention IDLE) and not `usage/python/README.md` (which does mention IDLE) !

@JamesH65 @bennuttall is it worth trying to restructure (or merge) those pages to prevent the duplicated content?

JamesH65, 2019-07-01:
I tend to avoid any of the Python pages as not particularly well versed....

commit 3ecc70b3 (2019-07-01): Update python software guide, close #1191