raspberrypi/documentation #1192: Add ID's to every answer/question (h3) in the FAQ

Janghou, 2019-07-02:
Adding an ID to every question h3, making direct linking and scrolling possible.
 
terse

    #wine
    #64-bit

or verbose

    #will-it-run-wine-or-windows-or-other-x86-software
    #the-processors-on-the-latest-pi-models-are-64-bit-but-i-cannot-find-an-official-64-bit-os

That will ease linking to questions

JamesH65, 2019-07-02:
I thought about doing this, but it was just too much of a PITA at the time, and time is short. I'm happy to accept a PR.

lurch, 2019-07-02:
If somebody does submit a PR for this, is there a small risk that external links to specific questions will then become 'broken' as other questions get removed/edited/added in future?

Janghou, 2019-07-02:
@lurch. It's the hash, so it will still link to the FAQ.  If the questions is removed, it doesn't scroll.
If the answer is edited, no problem.

And it's a FAQ, so questions can't change, because they were asked. ;)

@JamesH65, can't it be done automatically by the parser for the verbose version.
```
JS example
document.querySelectorAll("h3").forEach(
x=> x.id=x.textContent.toLowerCase().replace(/ /g,"-").replace(/[^-\w]/g,""))
```

epoch1970, 2020-11-06:
Created a PR #1724

JamesH65, 2020-11-06:
Fixed.

closed 2020-11-06 by JamesH65