raspberrypi/documentation #1199: Add PoE HAT documentation

JamesH65, 2019-07-12:
Currently little or no official information available.

6by9, 2019-07-12:
A grep on the repo for poe gives results
```
glossary/README.md:- **PoE** - Power over Ethernet, a mechanism to power a device through its Ethernet connection; we produce a [PoE HAT](https://www.raspberrypi.org/products/poe-hat) for the Raspberry Pi 3 Model B+ and Raspberry Pi 4 Model B.
faqs/README.md:- Is Power over Ethernet (PoE) possible?
faqs/README.md:Yes: if you own a 3B+ or 4B, you can use our [official Raspberry Pi PoE HAT](https://www.raspberrypi.org/products/poe-hat/) to power that Pi over Ethernet. For other models, there are adapters that split the voltage off the Ethernet line before connecting to the Pi. However, we have not tested any third-party PoE solutions, so we cannot recommend any of these. 
hardware/raspberrypi/mechanical/README.md:## Raspberry Pi PoE HAT
hardware/raspberrypi/mechanical/README.md:- [Board](rpi_MECH_PoEHAT.pdf)
hardware/raspberrypi/mechanical/rpi_MECH_4b_4p0.dxf:PoE
hardware/raspberrypi/conformity.md:| PoE HAT | [EU](./compliance/rpi_DOC_PoeHAT_EU.pdf)  &nbsp; &nbsp;[USA/FCC](./compliance/rpi_DOC_PoeHAT_FCC.pdf)| |
```
We appear to have all the other products listed under https://www.raspberrypi.org/documentation/hardware/, so that seems a logical place to add some more info.

mdlam92, 2020-01-20:
Any update on this?

JamesH65, 2020-01-21:
No.

lurch, 2020-01-21:
@mdlam92 Out of curiosity, what kind of info are you looking for in the PoE HAT documentation?

mdlam92, 2020-01-24:
@lurch I want to make a horrible amalgamation of a hat that combines this board with my board for gundam wing synergies

JamesH65, 2020-01-27:
This doc will be usage only, not technical details. Or at least, that was my original plan.

lurch, 2020-01-27:
I suspect @mdlam92 would be better served by https://pinout.xyz/ (which is unfortunately also [missing](https://github.com/Gadgetoid/Pinout.xyz/issues/343) the PoE HAT)

aallan, 2021-07-11:
The original PoE HAT is officially classified as "still in production, but not recommended for new designs". If we're doing any documentation updates for it, it'll be usage instructions only.

closed 2021-07-11 by aallan