raspberrypi/documentation #1201: Can't get RPI3B or 3B+ to boot from a USB HDD

ThomasFreedman, 2019-07-16:
I'm fairly confident this is a documentation issue.

I am attempting to boot a Raspberry Pi 3B (and 3B+) from a 2TB USB HDD. I have reviewed [this](https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/msd.md) and [this](https://www.raspberrypi.org/documentation/installation/installing-images/linux.md) and made several attempts on both models and I can't get it to work on either.

According to the first link above, it should be possible to boot those models from a USB HDD. The 3B requires a reprogram, which I performed and verified according to the directions. The first link above says you copy the image to the HDD the same way you do for an SD card.

If that is true wouldn't it be possible to pull a bootable SD card from the Pi's SD slot, place it in a USB CD card reader, insert that into one of the Pi's USB ports and boot from it that way? I tried it and that does _not_ work either, and if the doc I first linked was correct I'm not sure why it  wouldn't.

I'm using a Seagate HDD with a USB 2.0 SATA interface, and this Raspbian img: 2019-04-08-raspbian-stretch-lite.img.

JamesH65, 2019-07-17:
TBH, you might be better off asking this on the forums, where the problem can be determined, then I can update the docs to match whatever comes up.

I'll flag up the guy who knows about this though, just to see if he can help. @ghollingworth

ghollingworth, 2019-07-17:
Difficult to be sure why it is not working, in general the reasons why USB MSD doesn't work is because of power and start up time.  

Many USB HDD take many seconds to register on the USB bus and the Raspberry Pi only waits for 2 seconds, you can extend this timeout which can make some drives work.

In the past I've just used the SD clone application on a Raspberry Pi to clone its SD card to the USB HDD and that worked fine.

Just to check other things are working, plug ethernet into a switch and if you've sucessfully programmed the OTP then the LEDs will turn on the switch as it tries to boot from the network...

Another thing you can do is to get an SD card and copy _only_ the bootcode.bin file onto it.  This will use a slightly different version of the MSD code to read it (it also contains the UART debug output which you can enable)

Gordon

ThomasFreedman, 2019-07-17:
> Just to check other things are working, plug ethernet into a switch and if you've sucessfully programmed the OTP then the LEDs will turn on the switch as it tries to boot from the network...

So idea is remove SD card, plug in network (only ether or wifi too?), and green light would flash if OTP was programmed to boot USB? Already tested that by booting from a USB stick. Works for both 3B and 3B+

> Many USB HDD take many seconds to register on the USB bus and the Raspberry Pi only waits for 2 seconds, you can extend this timeout which can make some drives work.

Can you point me to docs on how to extend that timeout?

Not sure if rom space was reason, but it would make sense to look for HDD longer than 2 seconds of no SD card is present.

ghollingworth, 2019-07-18:
You can be sure about it by booting your Pi from an SD card and unplugging the HDD.

Then in a terminal do

`sudo tail -f /var/log/kern.log`

Then plug in the hard drive and count the seconds until it appears in kern.log

lurch, 2019-07-18:
> Can you point me to docs on how to extend that timeout?

https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/

JamesH65, 2019-09-18:
Closing, not a docs issue.

closed 2019-09-18 by JamesH65