raspberrypi/documentation #1202: Post RPi 4B Documentation Update

generaleramon, 2019-07-16:
as discussed in [this Thread](https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=245505), the raspberrypi/documentation is outdated/incomplete and some informations are missing after the RPi 4B release.

Examples: 
-SDRAM over-clocking not supported
-HDMI blanking not working at the moment
-gpu overclock not applying
-dinamic voltage, not working, i can change the voltage by setting both over_voltage and over_voltage_min at the same value
-etc etc.

generaleramon, 2019-07-17:
arm_freq_min=xxx - Defaut to 640Mhz, >640Mhz reverted to 640Mhz, 500Mhz rounded to 533Mhz, 

core_freq=xxx - Default to 500Mhz, up 600Mhz accepted, 625Mhz clock the freq to 250Mhz

gpu_freq=xxx - Default to 500Mhz, up to 600MHz accepted but only as "Turbo" clock, downclock to 500Mhz when "heavy" loaded

gpu_freq_min=xxx - Default to 500Mhz, >500Mhz reverted to 500Mhz, 31Mhz "Deep-Idle" Freq

JamesH65, 2019-07-17:
@popcornmix @timg236 Do these numbers look OK to you? What do we want to say about overclocking the Pi4?

timg236, 2019-07-17:
Fore the core-clock, except for 600 Mhz it's not supported and it's unlikely that we'll put any effort into making this work anytime soon.

JamesH65, 2019-07-17:
Someone has posted #1204 which I think covers much of this.

timg236, 2019-07-17:
The only caveat is that 600 MHz may require some over_voltage

The launch firmware applied some overvoltage at 600MHz with hdmi_enable_4k. In recent releases this has been replaced with hdmi_enable_4kp60 which actually changes some other clocks as well for lower power 4Kp60 and runs core at 550 MHz. This could be regarded as an easy over-clocking option because you can run the display at a lower resolution/refresh rate.

For over-clocking there is no automatic voltage boost and user's will have to experiment with this at their own risk (more likely to reset if not enough volts) and manage thermal issues.

generaleramon, 2019-07-17:
> The only caveat is that 600 MHz may require some over_voltage
>
> The launch firmware applied some overvoltage at 600MHz with hdmi_enable_4k. In recent releases this has been replaced with [...]

ok, good infos. Thanks
What about the "over_voltage"/"over_voltage_min" not working independently? I can change the voltage only if both are set to the same value, so if i want to overclock i will have a high idle voltage(wasting power and increasing temps) and if i want a nice and low idle voltage i can't overclock. I would really like to play around with this new 28nm chip

Askmewho, 2019-11-16:
Many many people are claiming gpu further than 600 GHz... Could someone clarify this topic. Because I cant go higher than 600 (as expected)  but some users, even respectable forums users are claiming that.

JamesH65, 2019-11-16:
Some chips will go faster, some won't. We only guarantee stock settings. You overclock at your own risk.

Askmewho, 2019-11-16:
No, you dont undersrand, ive said than its impossible to go over 600 mhz....or not? Because every one its going further and there is no performance gain because treadx or the eeprom implementation its not allowing that. My pi can boot on 750 oc on gou, but its not gaining any performance boost, also, i cant see the core going higher.

Askmewho, 2019-11-16:
In the documentation of config.txt, it says than 600 mhz its the limit... its or its not.  Why you are soo unclear with that, people are putting in risk his hardware for no reason. You should be clear here. Thanksss

JamesH65, 2019-11-16:
So, any reason you don't believe the official documentation? You have said you can set the config entry to >600, but it doesn't actually set it >600. Other people try at set >600 but see no performance gains, so  it sems that, as stated in the docs, that the limit is 600, and even if you set it to more it is clamped to 600. Just to confirm, @timg236 Is that correct?

This is a clocking thing, nothing to do with threadx or eeprom, and no hardware is at risk.

Askmewho, 2019-11-16:
How? Who limit your oc capabilities if its not threadx and the eeprom firmware?

timg236, 2019-11-16:
The max supported value is 600. Above that may or may not work or might get ignored. This is unlikely to change and would not make a big performance difference so suggest closing this issue

JamesH65, 2019-11-17:
> How? Who limit your oc capabilities if its not threadx and the eeprom firmware? Because it may be outdated...peopleclaims than new firmwares unlock the gpu over than 600 mhz... even very respectable forums.

1. Don't believe everything you read on the internet. Our official docs are generally correct, and if they are not, we fix them quickly.
2. Clocking is done using HW clocks. There are limits on those clocks - changing the firmware or the threadX OS cannot change the HW.

closed 2019-11-17 by JamesH65