raspberrypi/documentation #1205: getopt (util-linux) doesn't process  :: properly according to docs

ThomasFreedman, 2019-07-18:
There seems to be a problem with getopt (until-linux) processing optional parameters in the 4-8-2019 Raspbian Lite OS. 

According to [Debian docs](https://manpages.debian.org/stretch/util-linux/getopt.1.en.html), a double colon should be used to specify an optional parameter to an option, but instead it swallows the next option. Doesn't work as documented if I move the g:: to last position in invocation string either.

Here is invocation line: `getopt -o adfg::m:wh --long help -n "$0" -- "$@"`

lurch, 2019-07-19:
This sounds like a question for the forums, rather than the documentation repo?

JamesH65, 2019-07-19:
Indeed. Please post your question on the forum.

closed 2019-07-19 by JamesH65

ThomasFreedman, 2019-07-19:
Curious why you say that given the behavior described in the docs. I'll go ahead and post this there too, perhaps the Raspbian version of the util-linux package differs from Debian, but I find that doubtful. If they are the same, it's either documentation issue or a code issue, and which one is simply a matter of a decision based on consideration of the impact on code in the field. 

If this is reported as a bug by many in the field, probably should be logged as a code bug. On the other hand if that is rare, should be a change to docs, tho I don't know how such Debian / Raspbian issues are handled. Just noticed the discrepancy between Debian docs and Raspbian behavior.

lurch, 2019-07-20:
The link you posted in your first comment is to the Debian docs, whereas this repo is solely concerned with the documentation at https://www.raspberrypi.org/documentation/ which as you'll see never even mentions getopt.
If it's a documentation bug, then you'll need to open an issue upstream with Debian; whereas if it turns out to be a Raspbian-specific issue (i.e. a bug that only appears in Raspbian and doesn't appear in x86 Debian) then you could open an issue at https://github.com/RPi-Distro/repo

lurch, 2019-07-20:
In fact, I just looked into it a bit further and
```bash
/usr/share/doc/util-linux/examples/getopt-parse.bash  -a par1 'another arg' --c-long 'wow!*\?' -cmore -b " very long "
```
gives identical output on both Ubunutu on my laptop and Raspbian on my Pi.