raspberrypi/documentation #1209: Raspberry Pi 4B SD host controllers need documentation

Googulator, 2019-07-22:
The Raspberry Pi 4 has a new SD controller, and as a result, the SD host GPIO mappings appear to have changed.
Please document how the SD controllers can be routed to the various SD interfaces on the Pi 4 (GPIO 22:28, 34:40 and 48:54).

JamesH65, 2019-07-22:
Is there a page that already covers the older devices? If so please post a link to it, that'll save some time.

Right now I have no time to do much about this, so don't expect this to be done quickly.

lurch, 2019-07-22:
Does https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2711/rpi_DATA_2711_1p0_preliminary.pdf provide the info you're looking for?

Googulator, 2019-07-22:
No, this only covers GPIOs up to #27. What I'm looking for is how to select which controller gets routed to the SD card slot (GPIOs 48 to 53 on older Pis, not sure if that still holds for the Pi 4)

lurch, 2019-07-22:
> What I'm looking for is how to select which controller gets routed to the SD card slot

Out of curiosity, why do you need / want to change that?

Googulator, 2019-07-22:
I'm trying to bring up EDK2 on the Pi 4, but had no luck with the new emmc2 controller, so I'm trying to use one of the other controllers that EDK2 already has drivers for.

lurch, 2019-07-22:
A quick search for EDK2 finds https://github.com/tianocore/edk2

So I guess you need to switch which SD host controller is in use _before_ your bootloader gets loaded? @pelwell is this a question for the https://github.com/raspberrypi/firmware/ repo?

Googulator, 2019-07-22:
No, EDK2 can switch controllers just fine; it already does that on the Pi 3. The problem is, I can't find a way to route either of the older controllers (SDHOST or SDHCI) to the physical SD slot.

I'm starting to suspect the SD slot is no longer wired to GPIOs 48-53 on the Pi 4, but instead to dedicated SD pins.

pelwell, 2019-07-22:
> I'm starting to suspect the SD slot is no longer wired to GPIOs 48-53 on the Pi 4, but instead to dedicated SD pins.

You are correct. The new SDHCI-compliant EMMC2 interface doesn't appear on the GPIOs - it has dedicated pins, but to allow booting from SD card without a completely new boot ROM it is possible to map the old ARASAN/EMMC/SDIO interface to those dedicated pins. This is controlled by bit 1 of 0x7e2000d0 - 0=EMMC2, 1=legacy EMMC.

lurch, 2019-07-22:
The `pins_4b` section of https://github.com/raspberrypi/firmware/blob/master/extra/dt-blob.dts shows which GPIO pins are in use (and for what functions).

Googulator, 2019-07-22:
Does the "sdhost" controller still exist in any form on the Pi 4?

pelwell, 2019-07-22:
Yes - it's still accessible on GPIOs 22-27 (and 48-53, but that's not much use to you...)

Googulator, 2019-07-23:
> This is controlled by bit 1 of 0x7e2000d0 - 0=EMMC2, 1=legacy EMMC.

Any documentation for this MMIO register?

pelwell, 2019-07-23:
No, not yet. The only other relevant bit would be the one that enables the RGMII PHY signals onto GPIOs 46+, but you don't want to change that.

andreiw, 2019-07-26:
...apologies for being a little thick, but I don't see from pins_4b which pins the WLAN is wired up... presumably before switching bit 1 0x7e2000d0 to use Arasan instead of EMMC2, I'd have to tristate the existing connection to the WLAN?

lurch, 2019-07-31:
> ...apologies for being a little thick, but I don't see from pins_4b which pins the WLAN is wired up... 

That's not in `pins_4b`, but in the regular DeviceTree stuff - see https://github.com/raspberrypi/linux/blob/rpi-4.19.y/arch/arm/boot/dts/bcm2838-rpi-4-b.dts#L76 and https://github.com/raspberrypi/linux/blob/rpi-4.19.y/arch/arm/boot/dts/bcm283x.dtsi

> presumably before switching bit 1 0x7e2000d0 to use Arasan instead of EMMC2, I'd have to tristate the existing connection to the WLAN?

That's a question for @pelwell :wink:

pelwell, 2019-07-31:
> tristate the existing connection to the WLAN?

Can you unpack this a bit? At what time are you thinking of changing the mux, and what would be using the WiFi at the point?

lurch, 2019-11-06:
@andreiw @Googulator Do you have any further questions here, or can this be closed?

hvenev, 2020-05-27:
I have a question. Where can I find the documentation of the register at 0x7e2000d0?

pelwell, 2020-05-27:
We've published all we can at this time. This thread contains the best information available.

JamesH65, 2020-11-19:
All questions that could be answered are answered, so closing.

closed 2020-11-19 by JamesH65