raspberrypi/documentation #1217: Raspberry Pi 4B<->3B birds eye differences

andreiw, 2019-07-26:
Folks,

Trying to get the ol' Pi3 TianoCore UEFI port going on my 4GB board, and the GET_ARM_MEMORY (0x00010005) command via the mailbox  property interface insist I have the same amount as my 1GB Pi3.

...has the GET_ARM_MEMORY tag been deprecated in favor of something else? Has it been extended to return multiple ranges? How is the RAM laid out from the CPU's and GPU's perspective?

Also, it appears that for DMA purposes the shift for VC peripherals is still 0xc0000000. Can VC peripherals address the whole 4GB?

Can PCIe RC be made cache coherent?

JamesH65, 2019-07-26:
@pelwell Any comment?

pelwell, 2019-07-26:
> ...has the GET_ARM_MEMORY tag been deprecated in favor of something else?

Yes and no - the Pi 4 effort so far has been put into Device Tree support, but I know that some applications still use the mailbox interfaces.

> Has it been extended to return multiple ranges?

No, not yet.

> How is the RAM laid out from the CPU's and GPU's perspective?

There is a single contiguous block of RAM. Both the ARMs and the "GPU" see that at physical address 0, but the GPU only sees the first 1GB with the upper bits indicating cacheability. The GPU code and data is positioned at the top of that 1GB, leading to a hole in the ARMs' usable RAM.

> Also, it appears that for DMA purposes the shift for VC peripherals is still 0xc0000000.
Yes, because they are usually configured to bypass the GPU cache and go direct to RAM.

> Can VC peripherals address the whole 4GB?

There are 4 40-bit DMA channels with almost the same view of system memory as the ARMs that could be described as VC peripherals. The other 40/64-bit capable peripherals are intended for the ARMs and are unlikely to be accessed from the GPU after booting.

> Can PCIe RC be made cache coherent?

No.

andreiw, 2019-07-26:
Thank you @pelwell! What about the region below 4GB where MMIO lives? Does RAM wrap around that, or is that 64MB just lost?

pelwell, 2019-07-26:
It's lost, unless we eventually make the changes needed to allow the MMIO to be pushed up into the area above 16GB.

andreiw, 2019-07-31:
Thanks.

In the DTS I see a xhci@7e9c0000 block. This is not the discrete USB chip, but a built-in xHCI controller that you're not using, right? Any chance you could share the IO mux info to route it out to GPIO? (I also note there are no power/clock dependencies listed)

andreiw, 2019-10-01:
Any update on what the  xhci@7e9c0000 block is or how to use it?

pelwell, 2019-10-01:
@P33M - care to explain?

P33M, 2019-10-01:
The unimaginatively named `otg_mode` config.txt parameter does the necessary fiddling. Set to 1 to enable the xhci controller in place of dwc_otg, using the same USB2.0 PHY. Changing the controller over needs to be done at boot time as access to the other peripheral is explicitly disabled by the switch.

The otgid pin is left floating, so the PHY will stay in host mode.

andreiw, 2019-10-01:
Thank you. Is that a VideoCore device as well? So subject to the 1GB DMA limit and bus address translations?

andreiw, 2019-10-02:
I see that the Pi 4 still has the "legacy" CPU interrupt controller, but does it have the matching cascaded ("GPU" interrupt controller?) - the thing that was at 0x3f00B000 on the Pi 3? If it does, are the interrupt vectors documented anywhere?

lurch, 2019-10-02:
> The unimaginatively named otg_mode config.txt parameter does the necessary fiddling.

@P33M Would you mind adding that to an appropriate place somewhere under https://github.com/raspberrypi/documentation/tree/master/configuration/config-txt ?  Or is it too special-purpose to warrant documenting?

burtyb, 2019-10-02:
> > The unimaginatively named otg_mode config.txt parameter does the necessary fiddling.
> 
> @P33M Would you mind adding that to an appropriate place somewhere under https://github.com/raspberrypi/documentation/tree/master/configuration/config-txt ? Or is it too special-purpose to warrant documenting?

I'd like to see what's needed beyond "otg_mode=1" documented somewhere, especially if the xHCI host supports more devices than DWC OTG.

andreiw, 2019-10-03:
When otg_mode=0, I get a crash accessing the XHCI registers. Is there a way of /programmatically/ detecting whether access to xHCI is safe?

When otg_mode=1, I can detect and configure the xHCI controller, yet there is no device detected (whereas in otg_mode=0, the dwc2 driver *does* detect the device). What am I missing? According to the DT, this device is capabe of DMA access to RAM without any bus address translation.

Was I supposed to do any clock control or power state commands via the mailbox interface?

```
XhcCreateUsb3Hc: Capability length 0x20
XhcCreateUsb3Hc: HcSParams1 0x1000140
XhcCreateUsb3Hc: HcSParams2 0xC0000F1
XhcCreateUsb3Hc: HcCParams 0x220FE6D
XhcCreateUsb3Hc: DBOff 0x480
XhcCreateUsb3Hc: RTSOff 0x440
XhcCreateUsb3Hc: UsbLegSupOffset 0x0
XhcCreateUsb3Hc: DebugCapSupOffset 0xFFFFFFFF
XhcSetBiosOwnership: called to set BIOS ownership
XhcResetHC!
XhcInitSched:DCBAA=0x38E26000
XhcInitSched: Created CMD ring [38E26240~38E27240) EVENT ring [38E27240~38E29240)
InstallProtocolInterface: 3E745226-9818-45B6-A2AC-D7CD0E8BA2BC 39341038
XhcDriverBindingStart: XHCI started for controller @ 393F2C98
InstallProtocolInterface: 240612B7-A063-11D4-9A3A-0090273FC14D 3933D0A0
XhcGetCapability: 1 ports, 64 bit 1
UsbRootHubInit: root hub 3933CC98 - max speed 3, 1 ports
UsbBusStart: usb bus started on 393F2C98, root hub 3933CC98
InstallProtocolInterface: 240612B7-A063-11D4-9A3A-0090273FC14D 3933B020
UsbRootHubInit: root hub 3933BA98 - max speed 2, 1 ports
UsbBusStart: usb bus started on 3A030C98, root hub 3933BA98
```

P33M, 2019-10-03:
The device tree reflects the state of the hardware. The firmware explicitly disables the node with the "usb" label so you should have neither a dwc2 nor dwc_otg compatible node if the xhci controller is selected.
Do not access the disabled hardware - the register bus has been physically multiplexed into a different peripheral.

When you say "no device connected", what are you trying to connect to/between?

burtyb, 2019-10-04:
> When you say "no device connected", what are you trying to connect to/between?

For me in Raspbian with otg_mode=1 the xHCI host is there but a mass storage device connected to the USB-C (externally powered) which works with dwc2 doesn't show up or show anything in the logs when connected.

lurch, 2019-10-04:
Is it anything more than curiosity that is causing people to want to use xhci instead of dwc2?

pelwell, 2019-10-04:
It should be higher performance with lower overhead, but lacking the OTG capabilities.

pelwell, 2019-10-04:
And it ought to go without saying, you may get some assistance but this is an unsupported use case...

andreiw, 2019-10-04:
Thanks @P33M...this is inside a UEFI implementation for Pi 4. Is there a register I can check to figure out which controller is used? Some pinmux routing? A way without device tree. The way UEFI boots DT is optional (passed via config.txt) so I’m looking at hw method...

Right now the UEFI impl (derived from Pi 3 one) uses DWC2 driver via Type C. I wanted to get the xHCI USB2 working before trying to get the PCIe up. I connect a Type A USB hub using an A<>C adapter. This arrangement works with DWC2.

I do see vbus, but still no events from xHCI... looks like something else is needed to be done beyond just otg_mode.

A

JamesH65, 2019-11-06:
This is clearly not a documentation issue (or at least, it not really something we would put in our documentation because of its niche use case), and should really be discussed on the forums instead of here, so closing.

closed 2019-11-06 by JamesH65