raspberrypi/documentation #1225: Etcher requires zenity on Linux

alexlemaire, 2019-07-31:
## Issue :boom: 

Was unable to flash the latest [Raspbian Buster Lite image](https://www.raspberrypi.org/downloads/raspbian/) on my SD card with [balenaEtcher](https://www.balena.io/etcher/).

## Steps to reproduce :computer: 

1. Download and extract balenaEtcher appImage on a Linux machine
2. Download the latest Raspbian Buster Lite image from the official website
3. Start balenaEtcher
4. Select the Raspbian Buster Lite image archive and your SD device and try to "Flash!"
5. Enter your password in the SUDO prompt

This should then throw an error where something unexpected seem to happen when trying to flash.

## Fix :wrench: 

After some Googling ([see here](https://linuxhint.com/install_etcher_linux/)), I realized that the system needs to have _zenity_ installed to be able to flash an SD card via balenaEtcher.

## Update request :question: 

Would be nice to **include a note about that for Linux users** to avoid troubles for them.

I would gladly fork and open a pull request to handle this if this issue is approved! :+1:

JamesH65, 2019-07-31:
Feel free to provide a pull request. I would suggest saying that zenity _might_ need to be installed if not present

lurch, 2019-07-31:
I think this might count as an Etcher problem, rather than a Raspberry Pi documentation issue? Try creating an issue [here](https://github.com/balena-io/etcher/issues).

alexlemaire, 2019-08-01:
Alright, I created a pull request here with the update but will also open an issue on Etcher side so they can get that fixed.
In the meantime I think it may be worth having the note inside of Raspberry Pi documentation just to avoid problems for users. This can be removed as soon as Etcher team found a workaround.

JamesH65, 2019-11-06:
Change merged. #1230

closed 2019-11-06 by JamesH65