raspberrypi/documentation #1227: Glossary still not linked

lurch, 2019-07-31:
As I mentioned [here](https://github.com/raspberrypi/documentation/issues/801#issuecomment-500826570) there's still nothing linking to the glossary, so the _only_ way to find it is by directly typing in https://www.raspberrypi.org/documentation/glossary/
Opening an issue so that it doesn't get forgotten about...

Also, should the glossary mention EEPROM?

JamesH65, 2019-08-07:
Just added some text the documentation landing page to link to the glossary and the FAQ,  but not sure if that part is in github as its the one with the icons so is obviously modified somewhere. Will see in an hour or so!

lurch, 2019-08-07:
https://github.com/raspberrypi/documentation/blob/master/README.md has had a "Technical FAQ" link  on it for over a year, but that's never appeared on https://www.raspberrypi.org/documentation/ so I assume the https://www.raspberrypi.org/documentation/ index page is hand-edited / manually-updated ?

JamesH65, 2019-08-07:
I'm beginning to suspect the same.. will try and find out stuff.

andrum99, 2019-09-08:
https://www.raspberrypi.org/documentation appears to have originally been based on the github version, since adding README.md to the end gets the same content as without.  However, it has seems to have been hand-edited since then and excluded from whatever process copies everything over from the github repo.

In the absence of a web team contact that I know about, pinging @bennuttall, since he proposed a solution in https://github.com/raspberrypi/documentation/issues/801#issuecomment-500831234

JamesH65, 2019-09-09:
I've spoken to @LizUpton re: this, and we are working on a solution. Not sure of timescales. I'd really like to be able to edit that page!

JamesH65, 2020-11-19:
Still not linked so leaving this open for the moment. No idea what is going to happen here.

aallan, 2021-06-07:
The glossary will probably be removed — or at least heavily truncated — as part of #1911. But in any case this issue will be resolved as part of the update

closed 2021-06-07 by aallan