raspberrypi/documentation #1228: Problem with URLs ending with brackets

lurch, 2019-07-31:
As I mentioned [here](https://github.com/raspberrypi/documentation/pull/1186#issuecomment-512287987) and [here](https://github.com/raspberrypi/documentation/pull/1208#issuecomment-513811950) there's a problem with https://www.raspberrypi.org/documentation/ displaying URLs that end with a `)` character.
Opening an issue so that this doesn't get forgotten about...

andrum99, 2019-09-08:
According to https://en.wikipedia.org/wiki/Percent-encoding, parentheses are reserved characters in RFC3986 - Uniform Resource Identifier (URI): Generic Syntax. This means they are meant to be escaped. The mapping is:

( -> %28
) -> %29

lurch, 2019-09-18:
Fixed, see #1278

Thanks @andrum99 :+1:

closed 2019-09-18 by lurch