raspberrypi/documentation #1234: Possible incorrect cmdline.txt flag

OutsourcedGuru, 2019-08-01:
> plymouth.ignore_serial_console: tells Plymouth not to output a boot splash on any serial console which may be configured.

Having done plenty of research, I would suggest that this flag is plural rather than singular. And furthermore, there appears to be some confusion over the use of underscores versus hyphens in this flag.

This seemed to work as expected with Stretch using the singular version with underscores. After the upgrade to Buster it seems like undervoltage warnings from `dmesg` were clobbering the custom splash screen.

By changing my `/boot/cmdline.txt` to the plural version and appending `fbcon=map:2` it's now behaving again with Buster.

I reference the [main.c file](https://gitlab.freedesktop.org/plymouth/plymouth/blob/master/src/main.c#L2259) in the source code of plymouth. It appears to be in plural and possess hyphens rather than underscores.

What's difficult is that there are many competing suggestions for setting up plymouth on Raspbian. Perhaps someone can look into this and as necessary, update the documentation here.

andrum99, 2019-08-02:
Per https://gitlab.freedesktop.org/plymouth/plymouth/commit/6e50233b031ed301813de7f075ae0d606eae010c

`plymouth.ignore-serial-consoles` has always been the correct name of this option. The error in our docs comes from when that page was originally written - it seems nobody noticed until now, so good catch.

What does `fbcon=map:2` do? I've read the relevant section of https://www.kernel.org/doc/Documentation/fb/fbcon.txt and I don't really understand what it is saying.

OutsourcedGuru, 2019-08-02:
Bringing that section here:
```
fbcon=map:<0123>

        This is an interesting option. It tells which driver gets mapped to
        which console. The value '0123' is a sequence that gets repeated until
        the total length is 64 which is the number of consoles available. In
        the above example, it is expanded to 012301230123... and the mapping
        will be:

		tty | 1 2 3 4 5 6 7 8 9 ...
		fb  | 0 1 2 3 0 1 2 3 0 ...

		('cat /proc/fb' should tell you what the fb numbers are)

	One side effect that may be useful is using a map value that exceeds
	the number of loaded fb drivers. For example, if only one driver is
	available, fb0, adding fbcon=map:1 tells fbcon not to take over the
	console.

	Later on, when you want to map the console the to the framebuffer
	device, you can use the con2fbmap utility.
```

Using a single number 2 rather than a list presumably repeats `2222222` as the one you want in all cases. The reference to this suggestion comes from Raspberry Pi StackOverflow [here](https://raspberrypi.stackexchange.com/questions/59310/remove-boot-messages-all-text-in-jessie) by **Mich** on Oct 18 '18 at 20:18 as a comment.

Presumably it maps terminals to framebuffers. Since Plymouth works using them, it's somehow involved.

OutsourcedGuru, 2019-08-02:
I'm trying to find where that was.

I think you have the correct file in question. [Note](https://github.com/raspberrypi/documentation/pull/1068)

For what it's worth, I'm not convinced the blurb description next to this flag in the documentation is at all accurate. The purpose of the flag is to prevent many of the console messages from bubbling up to the terminal that's associated with Plymouth.

andrum99, 2019-08-03:
> For what it's worth, I'm not convinced the blurb description next to this flag in the documentation is at all accurate. The purpose of the flag is to prevent many of the console messages from bubbling up to the terminal that's associated with Plymouth.

From what I can see it is half accurate - Plymouth won't output a bootsplash on a serial terminal, but what it will do is, as you say, prevent the console boot messages - service startup and the like - from appearing on any serial console.

andrum99, 2019-08-03:
See #1235

OutsourcedGuru, 2019-08-04:
Well done. (Thanks.)

lurch, 2019-08-05:
Fixed in #1235

closed 2019-08-05 by lurch