raspberrypi/documentation #1238: pv utility to visualize image extraction from ZIP file to SDHC card

PSLLSP, 2019-08-10:
Just a hint for documentation at 

https://www.raspberrypi.org/documentation/installation/installing-images/linux.md

This command is in the documentation just now:

`# unzip -p 2019-07-10-raspbian-buster.zip | sudo dd of=/dev/sdg bs=4M conv=fsync
`
It is possible to visualize copy progress by adding command ``pv`` to the pipe. There is high chance that ``pv`` utility is not installed by default and has to be added to the host with command ``apt install -y pv``.

`# unzip -p 2019-07-10-raspbian-buster.zip | pv | sudo dd of=/dev/sdg bs=4M conv=fsync`

Copy progress is visualized with nice progress bar!

lurch, 2019-08-10:
Or you could use the `status=progress` option of dd, which is already mentioned on that page? :slightly_smiling_face:

PSLLSP, 2019-08-10:
I missed that note, I have no idea that ``dd`` can do that. ``status=progress`` is a new option that was added to ``dd`` during the last 4 years... It means,I don't need ``pv`` anymore :-)

JamesH65, 2019-09-18:
Closing this issue as questions answered/issue resolved.

closed 2019-09-18 by JamesH65