raspberrypi/documentation #1247: Documentation not up-to-date with Buster - Install a firewall

Truesh00ter, 2019-08-16:
Regarding: [Install a firewall](https://github.com/raspberrypi/documentation/blob/master/configuration/security.md#install-a-firewall), when you would try to enable the ufw it fails with an error as described and solved here: [ufw and iptables on buster](https://raspberrypi.stackexchange.com/questions/100598/ufw-and-iptables-on-buster) a fairly simple solution, but it does indicate that the guide isn't up-to-date anymore with Buster. 

I don't think linking to that solution is the way to go here, and I don't know what to add to keep the guide correct, bear in mind I used it to find my way on Debian and on Raspbian.
Anyhow, would anyone be so kind to fix this in the official guide?

andrum99, 2019-09-08:
I tested this on a fully up-to-date install of Raspian Buster today and it runs ufw without any issues. I also have the exact same package versions as the user in that thread on stackexchange.com - only the kernel if different. I suspect a kernel update may have fixed this - it's definitely working now.

For reference, the Raspbian config of iptables selection is as follows:

```
pi@livingroom:/etc/default $ sudo update-alternatives --display iptables
iptables - auto mode
  link best version is /usr/sbin/iptables-nft
  link currently points to /usr/sbin/iptables-nft
  link iptables is /usr/sbin/iptables
  slave iptables-restore is /usr/sbin/iptables-restore
  slave iptables-save is /usr/sbin/iptables-save
/usr/sbin/iptables-legacy - priority 10
  slave iptables-restore: /usr/sbin/iptables-legacy-restore
  slave iptables-save: /usr/sbin/iptables-legacy-save
/usr/sbin/iptables-nft - priority 20
  slave iptables-restore: /usr/sbin/iptables-nft-restore
  slave iptables-save: /usr/sbin/iptables-nft-save

```

JamesH65, 2019-09-18:
Closing this issue as questions answered/issue resolved.

closed 2019-09-18 by JamesH65