raspberrypi/documentation #1249: SD card size limit bug in pre 2837B0 SoCs needs clarified

andrum99, 2019-08-17:
Some time ago, I raised a PR to have the following added to installation/sd-cards.md:

> Note: Only the Raspberry Pi 3A+, 3B+ and Compute Module 3+ can boot from an SD card larger than 256 GB. This is because there was a bug in the SoC used on previous models of Pi.

This was then added to the docs.

Recently someone on the forums pointed out that this may not actually be correct. Re-reading where I got the information from, it seems that I jumped the gun a bit. It seems that there are at least two possibilities:

1. The bug is indeed to do with the total addressable size of the SD card
2. The bug only applies to the size of a partition or partitions which are on the card.

For example, suppose a 512GB SD card is re-formatted with a FAT32 filesystem and NOOBS installed. The total size of the partition, and of the card itself, exceeds 256GB. So the Pi won't boot.

Now let's suppose that the card is re-partitioned with a 128GB FAT32 partition and NOOBS installed, and the rest of the card left blank, i.e. unpartitioned. If (1) above applies, the card will be unbootable. If (2) above applies, the card will successfully boot.

Filing this bug here so I don't forget about it.

lurch, 2019-08-18:
> Some time ago, I raised a PR to have the following added to installation/sd-cards.md:

For reference, this was  #1136

> It seems that there are at least two possibilities:

@andrum99 I _think_ https://github.com/raspberrypi/noobs/issues/540 might suggest that it's (2) ? (specifically the size of the first FAT `/boot` partition, since AFAIK that's the only part that's read by the bootrom in the SoC? After that, `bootcode.bin` and `start.elf` (and then `kernel.img`) take over.)
Although perhaps the "soft-reboot" used by NOOBS adds an extra layer of subtleness? https://github.com/raspberrypi/noobs/wiki/NOOBS-partitioning-explained#noobs-bootup-boot-mode

ping @ghollingworth for possible clarification

lurch, 2019-11-06:
Closing, as the it seems the person asking this has since deleted their GitHub account (?!?)

closed 2019-11-06 by lurch