raspberrypi/documentation #1251: Multi-level lists broken on official Pi website

andrum99, 2019-08-17:
This is one I discovered a while ago but I'm just getting round to adding it as a separate bug. Documentation pages which contain multi-level lists are currently subtly broken on the official Pi website (raspberrypi.org). The breakage applies only to list items where the indent level is a power of 3, excluding zero. In other words indent levels 3, 6, 9 etc are affected. For those indent levels, any list item will be rendered as if it belong to the below indent level. So level 3 list items are rendered as level 2 list items, level 6 list items as level 5 items, etc.

Compare and contrast the following pages:

https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/host.md
https://github.com/raspberrypi/documentation/blob/master/hardware/raspberrypi/bootmodes/host.md

Rendered on github, the page shows the correct layout of multi-level lists. On raspberrypi.org it is broken as described above.

Interestingly, github.com seems to know about the possible breakage: if you go into edit mode it  renders 3rd, 6th etc level list items in red - see https://github.com/raspberrypi/documentation/edit/master/hardware/raspberrypi/bootmodes/host.md

andrum99, 2019-08-17:
Also affected by bug: https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/net.md

andrum99, 2019-08-17:
Note that there is nothing wrong with the markdown on these pages, the bug is in whatever is converting the markdown to HTML.

andrum99, 2019-08-17:
And another one: https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/bootflow.md

lurch, 2019-08-18:
@andrum99 I wonder if this might be related to https://github.com/Python-Markdown/markdown/issues/3 ?

andrum99, 2019-08-18:
> @andrum99 I wonder if this might be related to [Python-Markdown/markdown#3](https://github.com/Python-Markdown/markdown/issues/3) ?

I can't tell from that bug what is actually going from in that case. I've modified one of the wonky pages to apply their suggested fix - make all indents in the markdown 4 spaces instead of 2. Can you possibly have a look @lurch , and it it looks OK push it to master? We'll only know once it appears on raspberrypi.org if it has worked or not.

andrum99, 2019-08-18:
I suspect it won't help, but that's only based on the fact that github.com's editor still shows the defective lines in red.

andrum99, 2019-08-18:
OK, that was a useful guess as it was a potentially easy fix to the broken pages. This will need some investigation. How do the pages in github find their way onto the website?

lurch, 2019-08-18:
@andrum99 https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/host.md appears to have updated - does it look correct now?

andrum99, 2019-08-19:
> @andrum99 https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/host.md appears to have updated - does it look correct now?

Yes, it is now correctly rendered 😁

lurch, 2019-08-19:
Great! Feel free to submit a PR which applies the same fix to any other indented-lists you find, and then I can close this issue.

lurch, 2019-09-16:
@andrum99 Have all the relevant multi-level lists been fixed now? If so, can this issue be closed?

PR #1253 (2019-08-18): attempted fix for multi-level list bug
Attempt to fix #1251 by making all list indents on this page 4 spaces instead of 2.
closed 2019-09-16 by andrum99