raspberrypi/documentation #1266: balenaEtcher and privacy/GDPR

roan, 2019-09-05:
According to #https://github.com/balena-io/etcher/issues/2497 there are some concerns about privacy and GDPR compliance. As balenaEtcher is the suggested program for creating the sd card, could this be a problem?

JamesH65, 2019-09-05:
Thanks for the heads up, I'm passing it on to our legal team.

lurch, 2019-09-05:
https://github.com/balena-io/etcher/issues/2599#issuecomment-504042827 says the Etcher team are looking into it too.

rradar, 2019-09-05:
This is **well known to** the **balena** team since **minimum august 2017** (https://github.com/balena-io/etcher/issues/2497#issuecomment-508932329)

They (balena) don't take any actions since eve - no matter how many heads ups (@JamesH65). :whale2:

Guess we just need to wait for the Greece Authority (Hellenic Data Protection Authority (HDPA)) to take action and wake up some people here. Last company they fined 5 weeks ago was a 150k€ :boxing_glove:...

JamesH65, 2019-09-05:
This is not the place to discuss it - we are not balena. Since no actual GDPR lawyers have commented, on any threads I have read, to say it's a definite infringement, for the moment we will assume it isn't.  If balena want to take the risk, that is down to them, but we won't be playing internet lawyers here.

closed 2019-09-05 by JamesH65

pinchies, 2020-03-07:
@JamesH65 People are upset about the privacy infringement, not just the GDPR infringement.

JamesH65, 2020-03-08:
Not sure what you point is, on a 6 month old closed thread. 

But, we have just released our own Imaging Software and changed out documentation to recommend that , so you don't really need to use Etcher if you don't want to.