raspberrypi/documentation #1277: Continuous Frequency carrier-wave for Wifi test EMC RBPi

Nislash, 2019-09-11:
Dear All, 

For an IoT Application on SpaceCraft, We have to test the Wifi 2.4GHz and harmonics that The Raspberry Pi Zero W could emit and the TxPower to certify that couldn't damage the spacecraft's RX component (a telecom spacecraft work like a big antena but at 36000 km for most of them). 

This EMC Test is not aviable :https://www.raspberrypi.org/documentation/hardware/raspberrypi/compliance/rpi_DOC_3bplus_EU_RED.pdf
Because is made for Human and not spacecraft.

We have to Set the RBPi (any model) Wifi modul to have Only the Carrier-frequency, no datamodulation, a Continuous Wave (CW).

We find any solution online or any modified Kernel Module (even a CONF_NL80211_TESTMODE=y does nothing) 

Does any one know a modified kernel, software, tool or bin such as the ESP32_RF_TEST.bin that set the ESP32 to that configuration (wifi frequency carrier only, no data, do frequency modulation)

Sincerly,

Stanislas Dedieu

JamesH65, 2019-09-11:
This isn't actually a docs issue as such, but I will pass it on to someone who may be able to help. Please can you email info@raspberypi.org, FAO James Hughes and we will hopefully be able to deal with this appropriately.

Nislash, 2019-09-11:
Thankss you very much for this quick answer ! email send. 
I apologize for having posting this in docs issue, we didn't know were to find help.

JamesH65, 2019-09-11:
No worries, you did at least reference a docs page in your issue!

Nislash, 2019-09-11:
We have find the solution, we ask to compliance@raspberrypi.org our issue (the same that here) and a roger thornton answer us and send us the specified file to set the RBPi as Wifi and BT continiuous Frequency mode.

Thanks you all for all the fast support you bring to us!!!! love

JamesH65, 2019-09-11:
Cool, hope it goes well.

closed 2019-09-11 by JamesH65