raspberrypi/documentation #1291: improve docs on SD card partitioning boot requirements

GeoffWilliams, 2019-09-22:
The Raspberry PI has some boot requirements around SD card partitioning that do not appear to be documented. Testing with a Raspberry PI 3B, I found:

1. Pure GPT partition tables are not bootable. Raspberry PI will only boot SD card with an MBR or hybrid MBR partition table
2. Only the **first** partition will be scanned for `bootloader.bin` and this partition must be formatted `FAT`

If your SD card uses a hybrid MBR, you must ensure that the `FAT` partition is listed as the **first** MBR partition for the PI to be able to boot. Note that the order of partitions doesn't have to correspond to their physical order.

I'd like to add the above to the docs somewhere but I'm not sure on the right location?
* `hardware/raspberrypi/bootmodes/sdcard.md` seems like a good candidate but at the moment contains pretty basic info, not technical details like this
* `hardware/raspberrypi/bootmodes/bootflow.md` looks like a good place but this may apply to the PI4 too and there is a note at the start of the page about this only applying to BCM2837

If anyone can let me know the best place for this to live I'll put a PR together.

lurch, 2019-09-23:
It's not in the documentation repo, but a _long_ time ago I made some pages which you may find useful:
https://github.com/raspberrypi/noobs/wiki/Standalone-partitioning-explained
https://github.com/raspberrypi/noobs/wiki/NOOBS-partitioning-explained

The bootflow.md page you mentioned says that for the BCM2837 "It is no longer necessary for the first partition to be the FAT partition, as the MSD boot will continue to search for a FAT partition beyond the first one." so I guess the behaviour on Pi3 boards is _slightly_ different to previous models.

The Pi4 doesn't yet support USB or network boot-modes, but it will do with future firmware updates (at which point I'm sure the documentation here will be updated as necessary).

I think the information you're trying to document will only be useful to people trying to make their own custom OS images?
@JamesH65 Do you think that level of info belongs in the documentation repo? :man_shrugging:

GeoffWilliams, 2019-09-23:
Yeah that's exactly what I'm doing - making my own OS images based off buildroot. The noobs links have good info but I never thought to look there, although I did look at the noobs image and go "aha... its MBR".

The PI3 behaviour only jumped out at me when I started looking at other OS images I had the worked vs mine that didn't. As soon as I fixed the order the partitions were listed it booted right away.

lurch, 2019-09-23:
OffTopic: buildroot already has Raspberry Pi support, so I'd assume that it'd create the correct partitions for you automatically?

GeoffWilliams, 2019-09-23:
It does but I decided to see what happened if I started playing around with GPT and hybrid MBR - figured this should be written down somewhere since it took quite a while to find out these points

lurch, 2019-09-23:
"I decided to put diesel in my petrol car, to see what would happen"? :wink: :joy:

GeoffWilliams, 2019-09-23:
haha I see where your coming from. It did work though. Now I can boot hybrid MBR and also patched `sgdisk` to support this layout :rocket:

JamesH65, 2019-11-06:
Seem to me that the level of detail required here is much higher than the rest of the documentation, so it would be a bit out of place. Considering the usage is rather off the beaten track, I don't think we will be spending any time adding it. However, a PR would be considered.

closed 2019-11-06 by JamesH65