raspberrypi/documentation #1295: page listing bootable USB devices?

RichMorin, 2019-09-26:
The blog page [Pi 3 booting part I: USB mass storage boot beta](https://www.raspberrypi.org/blog/pi-3-booting-part-i-usb-mass-storage-boot/) lists several USB devices that are thought to boot without problems on the RasPi. The comments section includes several more.

Unfortunately, this information is unorganized and somewhat incomplete. Also, because the comments section is now closed, additions and corrections aren't possible. So, I would like to propose a page listing bootable USB devices. I'd divide it into sections by device type (e.g., flash drive, SSD). Each entry should contain the name, size, and model number for a device.

lurch, 2019-09-26:
Surely _any_ such list will always be incomplete? :man_shrugging: 
This suggestion reminds me a bit of https://elinux.org/RPi_SD_cards and I'm not sure the documentation repo is the place for such a list? @JamesH65 ?

JamesH65, 2019-09-26:
I agree with @lurch - lists of 'compatible' ancillaries are always out of date, no matter how often they are updated, and things might suddenly stop working anyway, so the testing burden is quite horrendous to keep this sort of thing up to date.

lurch, 2019-09-26:
> For example, is there a wiki where this could be located?

The link in my previous reply is a wiki... :wink:

RichMorin, 2019-09-26:
> For example, is there a wiki where this could be located?
> 
> The link in my previous reply is a wiki... 😉

Ah; I didn't realize that.  I've asked for an account and plan to
make a start on a page.

-r

JamesH65, 2019-09-27:
Going to close, as this sort of almost ephemeral  data isn't suitable for docs which cannot really afford to have inaccurate/very incomplete data in it.

closed 2019-09-27 by JamesH65