raspberrypi/documentation #1297: Error in your AP documentation

Memoi, 2019-09-28:
Hello,

your documentation https://github.com/raspberrypi/documentation/blob/master/configuration/wireless/access-point.md doesn't work !
Thanks to update it

lurch, 2019-10-02:
"It doesn't work" is never a helpful complaint :frowning_face: 

Please provide more details about exactly what doesn't work, what error messages you get, etc. etc.

Or if you're looking for more general help, ask on the [forums](https://www.raspberrypi.org/forums/)

Memoi, 2019-10-03:
Hi,

I found the problem, you forget to add :

sudo systemctl enable systemd-networkd

Without it the ip of br0 is 169.xxx so we loose ssh connection after the reboot because this service is not enable by default

lurch, 2019-10-03:
ping @XECDesign as I believe he looked at this most recently?

XECDesign, 2019-10-03:
I believe OP is right. It doesn't seem to be enabled by default.

It should be fixed when #1268 is merged.

jarrah31, 2019-12-08:
"sudo systemctl enable systemd-networkd" isn't in the instructions yet within "Using the Raspberry Pi as an access point to share an internet connection (bridge)". I came hunting on here for an answer as to why the IP I had specified wasn't being picked up, even after a reboot.

JamesH65, 2019-12-08:
https://github.com/raspberrypi/documentation/pull/1333 is still being copy edited. Sorry about the delay.

aallan, 2020-06-03:
#1333 got merged. Can we close this one?

JamesH65, 2020-06-03:
Guess so!

closed 2020-06-03 by JamesH65