raspberrypi/documentation #1298: Missing revision info?

andreiw, 2019-10-01:
Hi,

The  https://www.raspberrypi.org/documentation/hardware/raspberrypi/revision-codes/README.md is pretty thorough, but it doesn't document bit 31 being set.

What is bit 31 supposed to be?

I'm getting 80C03111 via the mailbox, which corresponds to a Pi4 4GB...but what's that 0x80000000 for?

A

P33M, 2019-10-02:
What is the full output as seen in /proc/cpuinfo?

JamesH65, 2019-11-06:
/proc/cpuinfo does not have the high bit set. 

Which mailbox call is returning with the high bit set?

lurch, 2019-11-06:
I'd assume @andreiw is talking about https://github.com/raspberrypi/firmware/wiki/Mailbox-property-interface#get-board-revision ?  I've not tested it myself though.
:man_shrugging:

JamesH65, 2019-11-06:
That bit appears to be the over voltage bit in the OTP - @andreiw Have you ever asked that bit to be set in config.txt?

You can double check what is on OTP using `vcgencmd otp_dump`

andreiw, 2019-11-06:
I’ve only set the never overvoltage option.

A

JamesH65, 2019-11-06:
Sorry, typo in my post, that is the never overvoltage bit, so that matches. 

I'll see whether I can add those bits to the docs.

JamesH65, 2020-11-19:
I believe this is fixed.

closed 2020-11-19 by JamesH65