raspberrypi/documentation #1301: After setting the bridge : systemd-networkd[146]: br0: Could not acquire DHCPv6 lease on NDisc request: Address already in use

Memoi, 2019-10-04:
Hello,

After setting the network [Bridge](https://www.raspberrypi.org/documentation/configuration/wireless/access-point.md)  I have a lot of errors about DHCPv6 lease.

I think it's because there is 2 network clients (dhcpcd and systemd-networkd) like they say [Here](https://bbs.archlinux.org/viewtopic.php?id=244690) , if I disable dhcpcd the error is gone but my RPI have only an iPv6 and not a v4. 

Thanks for your help

lurch, 2019-10-04:
This documentation will be changing anyway once #1268 has been copy-edited.

XECDesign, 2019-10-04:
Not sure, but OP's concern may still be valid with #1268. Might be a question for @epoch1970

epoch1970, 2019-10-04:
In #1268, there is no IP config in bridge mode. I suppose DHCPv6 won't be unhappy. DHCPv4 is happy.
The routed config is IPv4 only. I'm not really proficient in IPv6 in case a dual stack config is desirable :(

aallan, 2021-06-07:
Considering we're going to tighten up the documentation a lot after #1911, removing most of the material that isn't Raspberry Pi specific. This seems like something that is advanced enough that it's not going to get included in the streamlined documentation.

closed 2021-06-07 by aallan