raspberrypi/documentation #1311: Cross Compiling using Ubuntu 18.04 LTS from Windows 10 Store

WanaGo, 2019-10-22:
Hello,
I am trying to follow the instructions for cross compiling for the Pi4 on Windows 10 Ubuntu 18.04 LTS from the Windows Store.
The instructions was missing the 'make' package, which I had to install. But trying to execute:

make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- bcm2711_defconfig

results in:

$ make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- bcm2711_defconfig
arm-linux-gnueabihf-gcc: error trying to exec '/home/james/tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian/bin/../libexec/gcc/arm-linux-gnueabihf/4.8.3/cc1': execv: Exec format error
arm-linux-gnueabihf-gcc: error trying to exec '/home/james/tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian/bin/../libexec/gcc/arm-linux-gnueabihf/4.8.3/cc1': execv: Exec format error
arm-linux-gnueabihf-gcc: error trying to exec '/home/james/tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian/bin/../libexec/gcc/arm-linux-gnueabihf/4.8.3/cc1': execv: Exec format error
arm-linux-gnueabihf-gcc: error trying to exec '/home/james/tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian/bin/../libexec/gcc/arm-linux-gnueabihf/4.8.3/cc1': execv: Exec format error
arm-linux-gnueabihf-gcc: error trying to exec '/home/james/tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian/bin/../libexec/gcc/arm-linux-gnueabihf/4.8.3/cc1': execv: Exec format error
arm-linux-gnueabihf-gcc: error trying to exec '/home/james/tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian/bin/../libexec/gcc/arm-linux-gnueabihf/4.8.3/cc1': execv: Exec format error
init/Kconfig:17: syntax error
init/Kconfig:16: invalid option
scripts/kconfig/Makefile:104: recipe for target 'bcm2711_defconfig' failed
make[1]: *** [bcm2711_defconfig] Error 1
Makefile:534: recipe for target 'bcm2711_defconfig' failed
make: *** [bcm2711_defconfig] Error 2

I am not sure what I am doing wrong

Thanks

WanaGo, 2019-10-22:
I just started again, and got the same issue.

james@PC:~$ echo PATH=\$PATH:~/tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian-x64/bin >> ~/.bashrc
james@PC:~$ source ~/.bashrc
james@PC:~$ git clone --depth=1 https://github.com/raspberrypi/linux
Cloning into 'linux'...
remote: Enumerating objects: 66186, done.
remote: Counting objects: 100% (66186/66186), done.
remote: Compressing objects: 100% (61546/61546), done.
remote: Total 66186 (delta 5068), reused 17678 (delta 3715), pack-reused 0
Receiving objects: 100% (66186/66186), 177.77 MiB | 8.26 MiB/s, done.
Resolving deltas: 100% (5068/5068), done.
Checking out files: 100% (62346/62346), done.
james@PC:~$ cd linux
james@PC:~/linux$ KERNEL=kernel7l
james@PC:~/linux$ make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- bcm2711_defconfig
  HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/kconfig/conf.o
  YACC    scripts/kconfig/zconf.tab.c
  LEX     scripts/kconfig/zconf.lex.c
  HOSTCC  scripts/kconfig/zconf.tab.o
  HOSTLD  scripts/kconfig/conf
arm-linux-gnueabihf-gcc: error trying to exec '/home/james/tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian/bin/../libexec/gcc/arm-linux-gnueabihf/4.8.3/cc1': execv: Exec format error
arm-linux-gnueabihf-gcc: error trying to exec '/home/james/tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian/bin/../libexec/gcc/arm-linux-gnueabihf/4.8.3/cc1': execv: Exec format error
arm-linux-gnueabihf-gcc: error trying to exec '/home/james/tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian/bin/../libexec/gcc/arm-linux-gnueabihf/4.8.3/cc1': execv: Exec format error
arm-linux-gnueabihf-gcc: error trying to exec '/home/james/tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian/bin/../libexec/gcc/arm-linux-gnueabihf/4.8.3/cc1': execv: Exec format error
arm-linux-gnueabihf-gcc: error trying to exec '/home/james/tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian/bin/../libexec/gcc/arm-linux-gnueabihf/4.8.3/cc1': execv: Exec format error
arm-linux-gnueabihf-gcc: error trying to exec '/home/james/tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian/bin/../libexec/gcc/arm-linux-gnueabihf/4.8.3/cc1': execv: Exec format error
init/Kconfig:17: syntax error
init/Kconfig:16: invalid option
scripts/kconfig/Makefile:104: recipe for target 'bcm2711_defconfig' failed
make[1]: *** [bcm2711_defconfig] Error 1
Makefile:534: recipe for target 'bcm2711_defconfig' failed
make: *** [bcm2711_defconfig] Error 2
james@PC:~/linux$

I am unsure if its related or not, but I set my path to the x64 path, yet the error is showing the x32 path. I am running Windows 10 x64 on a 64bit machine, with this Ubuntu 18.04 LTS

Any ideas what I am doing wrong?

lurch, 2019-10-23:
> I am unsure if its related or not, but I set my path to the x64 path, yet the error is showing the x32 path.

Have you got **both** directories added to your path? I think the $PATH variable is searched in linear order, i.e. if $PATH is set to `/some/dir1:/other/dir2` and an executable named `prog` exists in both `/some/dir1` and `/other/dir2`, then running `prog` will run `/some/dir1/prog` (and `/other/dir2/prog` will be ignored).

WanaGo, 2019-10-23:
> Have you got **both** directories added to your path? I think the $PATH variable is searched in linear order, i.e. if $PATH is set to `/some/dir1:/other/dir2` and an executable named `prog` exists [...]

Hit the nail on the head, thats exactly what it was. Did the x32 one by mistake, then did the x64 and assumed it would overwrite it, but was a bad assumption. 
Thanks very much

lurch, 2019-10-23:
Can this be closed now then? :slightly_smiling_face: 

P.S. For any other issues like this in the future, you'd be much better off asking on the https://www.raspberrypi.org/forums/ first.

closed 2019-10-23 by WanaGo