raspberrypi/documentation #1316: login appears on incorrect screen

Fannattatic, 2019-10-26:
When Raspbian is booted with login desktop enabled the login prompt appears on HDMI-1 but in my case this isn't the primary screen, since HDMI-2 is the primary screen it should appear on HDMI-2.
I have included some attachments to show the displays and the screen configuration:

[image]
[image]
[image]

HDMI-1 resolution is 2560x1080 plugged into HDMI0
HDMI-2 resolution is 1680x1050 plugged into HDMI1

I don't think this will affect anything though

Also adding a button to make a specific display the default display would be very nice because currently the display to the left will be the default display.

JamesH65, 2019-10-27:
This is the wrong repo for this report, its should be in the raspbian desktop repo - this one is for documentation only.

@spl237 Which is the correct repo?

spl237, 2019-10-27:
The login prompt is always displayed on what X views as the primary monitor, which is HDMI-1 in a dual HDMI setup. Rearranging the displays, as you have done, does not affect which monitor X regards as primary.

It is possible to change the primary monitor by adjusting X settings, but because the only thing this actually affects is the location of the login prompt - it doesn't affect the taskbar, desktop, windows or anything else - the option to change it is not offered to the user, as it caused too much confusion as to what it actually did. If you want to change it yourself, you will need to manually reconfigure the xrandr settings in /usr/share/dispsetup.sh

JamesH65, 2019-10-28:
Closing this issue as questions answered/issue resolved. (And shouldn't really be in this repo anyway)

closed 2019-10-28 by JamesH65