raspberrypi/documentation #1324: Schematic Camera Module Rev 1.3

Bardo91, 2019-11-02:
I am trying to find the Schematics of the Camera module Rev 1.3. But it seems that it is missing. I can only find on the internet the Rev. 2.0 version. I am trying to replicate it by using the datasheet of the OV5647 camera (https://cdn.sparkfun.com/datasheets/Dev/RaspberryPi/ov5647_full.pdf). But it is not as clear as I expected how to use it to make it work with the raspberry pi.  Is it stored in some branch in the repo or hidden in the webpage?

JamesH65, 2019-11-02:
If it's not on our docs schematics page or on the camera , then we haven't released it. I've had a quick look,  and its does not appear to be available.

Bardo91, 2019-11-02:
Ok, thank you very much. I thought that is was released by Raspberry. Nevertheless, I found a hackaday project which uses the OV5647 to create a board with two cameras. It uses the right components for this camera. Here is the link (https://hackaday.io/project/167794-dual-camera-board-for-ov5674), just in case someone else needs it.

closed 2019-11-02 by Bardo91