raspberrypi/documentation #1325: vcgencmd get_mem arm result less on raspberry pi 4 with 4GB RAM

gwh-cpnet, 2019-11-05:
[image]

according to the [doc](https://github.com/raspberrypi/documentation/blob/JamesH65-patch-vcgencmd-vcdbg-docs/raspbian/applications/vcgencmd.md#get_mem), the sum of memories for the arm and gpu is supposed to be 4GB, right?

I have a stats script that runs on various types of raspberry pi. I hope I can get the memory specification right on all of them.

JamesH65, 2019-11-06:
This is expected behaviour (albeit rather counterintuitive). The firmware, which is what vcgencmd is interrogating, only sees the first 1G of RAM, so this call will return 1G - the gpu memory allocated.

To determine ARM memory size available to Linux, you should use Linux command, e.g. https://www.cyberciti.biz/faq/how-to-check-memory-debian-linux/

closed 2019-11-06 by JamesH65

gwh-cpnet, 2019-11-07:
@JamesH65  Thank you for the explaination.