raspberrypi/documentation #1327: Raspberry Pi 4 is missing from "USB device boot mode".

sirocyl, 2019-11-07:
**Page:** [documentation > hardware > raspberrypi > bootmodes > device](https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/device.md)

**Location:** List of devices which support USB Device boot mode.

The Raspberry Pi 4 seems to support this method of boot, according to [changes](https://github.com/raspberrypi/usbboot/commit/ecb8797287853acd0ac2ced40258f23a8990ceae) to the [USB boot code](https://github.com/raspberrypi/usbboot). I'm opening the issue here to verify that that is indeed the case, for correctness' sake.

A pull request will follow shortly.

lurch, 2019-11-07:
See also https://github.com/raspberrypi/usbboot/issues/40

@ghollingworth Is this officially supported, or is it still in a beta state?

JamesH65, 2019-11-08:
@ghollingworth Ping.

ghollingworth, 2019-11-17:
Currently there is no support for the mass storage device on Pi 4, so although it is possible to push the bootrom.bin / recovery.bin code into a Pi 4 through this interface, that code cannot continue any further.

JamesH65, 2019-11-18:
Closing this issue as questions answered/issue resolved.

closed 2019-11-18 by JamesH65