raspberrypi/documentation #1336: Adding Desktop Pager removes ability to switch/move/resize/close windows

scruss, 2019-11-16:
**Actions in desktop**:

1. right click in the menu bar, “Add/Remove panel items”

2. in Panel Preferences click “Add”, then select “Desktop Pager” and “Add”. 

3. Right-click on newly-created one-page pager,  and choose “Desktop Pager Settings”. Add a number (> 1)  in the Desktops tab.

4. Close settings.

**Expected result**: use desktop as normal, but with multiple workspaces.

**Actual result**: multiple workspaces are enabled, but the ability to switch/move/resize/close windows is lost. The only way to switch/manipulate windows is through right clicking the window indicator in the top bar.

This may be an issue with obconf writing unexpected values into the user's desktop configuration. Seems to be a recent issue, perhaps since Stretch. Didn't appear to be an issue around 2015. Not a stock/upstream LXDE user issue, either: I'm a long-time Lubuntu user and I can add many workspaces without problems. Seems also to be a problem on Raspberry Pi Desktop (for PC and Mac) too.

forum reference: https://www.raspberrypi.org/forums/viewtopic.php?f=116&t=251268 (probably in wrong forum)

XECDesign, 2019-11-16:
@spl237

spl237, 2019-11-16:
Multiple desktop shortcuts in openbox config were removed several releases ago, because we do not support the use of multiple desktops. They can be re-added by the user if desired, but that is at the user's own risk.

spl237, 2019-11-16:
Although, now I look at your forum post, it seems that it isn't just the desktop shortcuts that don't work with multiple desktops, but that something is fundamentally broken if windows can't be dragged. This is why we removed the shortcuts and don't install the pager by default; it's not a use case we support. I am not aware of any changes which we have made which would cause this problem, so it is almost certainly from upstream.

scruss, 2019-11-16:
The pager *is* installed by default: I can see it from “Add/Remove panel items” on a stock system. And desktop shortcuts do work, but they break the ability to move windows. I realize this isn't a priority fix, but since the newer Raspberry Pi models are viable desktop replacements, multiple desktops are going to be expected.

spl237, 2019-11-16:
The pager is not included in the default configuration of the taskbar. It is a standard part of the LXDE taskbar package, which is why it is present on the system; removing it completely would be unnecessarily tedious. Any plugins which are not included in the default Pi configuration are not supported. (Not least, because from experience, they tend to be buggy.)

Only a very small minority of users use virtual desktops; most find them unnecessarily confusing, and support for multiple monitors renders them unnecessary in most cases - as above, this is therefore not a priority. I will investigate to see if this is something broken by any of our changes, but if not, it won't be fixed by us.

scruss, 2019-11-18:
Further detail: as noted by user petr2 [here](https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=251268&p=1568003#p1567525), creating workspaces via `obconf` does not cause this problem to appear.

I use multiple workspaces because I have attention deficit/sensory problems, and one workspace per task allows me to focus.

JamesH65, 2019-11-18:
Just realised this is in entirely the wrong repo. Should be in the https://github.com/RPi-Distro/raspberrypi-ui-mods I reckon. I've tried to use transfer issue and that gives a 500 error from github...so not sure how to move it.

scruss, 2019-11-18:
I'll create one there and close this.
Now here: https://github.com/RPi-Distro/raspberrypi-ui-mods/issues/22

closed 2019-11-18 by scruss

XECDesign, 2019-11-18:
I don't think you can move issues across organisations, but it should've been opened here https://github.com/RPi-Distro/repo/issues/ (a catch-all for issues with packages in the raspberrypi.org repo)