raspberrypi/documentation #1346: Please do not recommend Etcher Spyware

sneak, 2019-11-26:
Wikipedia defines spyware as:

> Spyware is a software that aims to gather information about a person or organization, sometimes without their knowledge, and send such information to another entity without the consumer's consent. 

Etcher has done this for some time.  The developers' excuse for not changing this behavior is that they don't believe that they are breaking the law, as the data is anonymous (to them).  However, it includes IP address and timestamp information, which is more than sufficient to identify a user (and perhaps even their physical location), if not to them, then to the other people who gain access to this analytics data.

Regardless, this is not about what data is collected, simply that data is collected *without user consent or interaction*.  There is a setting to disable it, but by the time you can even get to the setting, the software has already leaked your data.

Etcher is **spyware**.  Please do not recommend spyware to your users.

sneak, 2019-11-26:
Their refusal to act can be found here: https://github.com/balena-io/etcher/issues/2057

JamesH65, 2019-11-27:
We have no intention at this stage of changing our recommendations. Until such time as Etcher is shown to contravene the GDPR things will likely remain as they are. Etcher provides a multiplatform and easy to use system for Raspberry Pi users, greatly simplifying creating of SD cards and the documentation required to support it.

closed 2019-11-27 by JamesH65

sneak, 2019-11-27:
Wikipedia defines spyware as:

> Spyware is a software that aims to gather information about a person or organization, sometimes without their knowledge, and send such information to another entity without the consumer's consent. 

## Required Elements

1. is software :white_check_mark: etcher is software
1. gather information about a person :white_check_mark: information: that the user is launching etcher
1. without their knowledge :white_check_mark: no indication is displayed at any time that this is happening
1. send information to another entity :white_check_mark: information is transmitted to LAN, ISP, interchange points, and hundreds of others
1. without consent :white_check_mark: no consent is asked for or given (and indeed, none exists)

This **silent** tracking **includes IP address and timestamp information**, which is more than sufficient to **identify a user (and perhaps even their physical location)** to the other people who gain access to this data, such as analytics providers, network hosts, interchange points, ISPs, and intelligence services (hi Ed!).

Upon opening Etcher for the **first time**, the following connections are attempted:

[image]
[image]

At no point am I prompted for consent, or provided the ability or UI to opt out.  This happens *silently*, regardless of user intent or consent.  Only after balena has been contacted by the software does the main application window open:

[image]

Then, the user could attempt to disable the settings, but by then their IP address (and physical location) has by this time already been transmitted to the manufacturer, likely against their consent and wishes.

This issue is not about the GDPR, or the legality of this collection, simply the very practical issue that the software phones home and **leaks the user's IP address to the developers and hundreds of others without consent or even notification**.  At no point does the user have the ability to disable this on first launch.  By simply phoning home, thousands of other people have gained access to the piece of information that a given user is using this software.

At present, that makes this application fit the definition of **spyware**.

Remember: humans have an inalienable right to privacy.  When this application leaks users' personal data (even if they do not save it, or don't receive it themselves - by *causing it to be sent out of their computer at all*) they have **infringed upon their human rights**.

Please do not recommend software to new computer users that abuses their human rights.

I fear for a generation of people growing up who become entirely accustomed to nonconsensual surveillance.  We can and should do better, especially on a teaching platform.

It has nothing to do with GDPR; it's a simple matter of ethics.  Etcher is unethical, abusive software that violates the consent of the user.

sneak, 2019-11-27:
It's literally only a matter of time before leaky tools like this get someone killed due to the data leakage.

JamesH65, 2019-11-28:
> It's literally only a matter of time before leaky tools like this get someone killed due to the data leakage of the phone-home hostnames.
> 
> We take usage of technology for granted in most safe and free countries, but platforms like the Pi are being used in a lot of ways and in a lot of places that don't immediately spring to our minds. This sort of spying is inconvenient for most, but _actively dangerous_ for some.

Doing well until the hyperbole.

neildmd, 2020-03-09:
JamesH65 Why is it such a big deal to have users ask to remove an unethical feature? Etcher was great until baleno took over! Please remove the remote calls or at least give us the option to disable this during install so it never leaks our IP.

JamesH65, 2020-03-09:
Why are you commented on a long dead thread? But since you ask.

1. Etcher is nothing to do with us except we have in the past recommended it, so we won't be changing any code in it. This is the Raspberry Pi docs issue tracker, not balena's.
2. We like proof, and ethicality is often in the eye of the beholder. Many people care not a whit about these 'leaks'.
3. We have already developed and now recommend our own Imaging program, so Etcher is history.

neildmd, 2020-03-09:
Sorry I should have checked the target, didn't realize this was rpi tracker. 
What is the imaging program you guys developed? Is it USBImager?

andreixk, 2021-08-13:
> Wikipedia defines spyware as:
> 
> > Spyware is a software that aims to gather information about a person or organization, sometimes without their knowledge, and send such information to another entity without the consumer's consent.
> 

That's not true.

Here's the actual definition of spyware from [wikipedia](https://en.wikipedia.org/wiki/Spyware):

"Spyware is software with ___malicious behavior___ that aims to gather information about a person or organization and send it to another entity in a way that ___harms the user___."

So, maybe stop lying to fit your agenda?