raspberrypi/documentation #1351: Need "zlib1g:i386" and "bc" for cross-compiling kernel

hhaoyan, 2019-11-30:
Problem: when I tried the cross-compiling instructions at (https://www.raspberrypi.org/documentation/linux/kernel/building.md), I found two more dependencies are required to build:
```
sudo apt-get install zlib1g:i386 bc
```

`zlib1g:i386` is for the cross-compiler (cross-compiler is i386 and I'm running x86_64) and `bc` is for the build process.

lurch, 2019-12-02:
That page has a section about "If you are on a 64-bit host system" ... ?

hhaoyan, 2019-12-02:
I see. Thank you for pointing out, but I guess we still need to add installation instructions for bc?

JamesH65, 2019-12-02:
Presumably just the following?

`sudo apt install bc`

lurch, 2019-12-02:
@JamesH65 I just remembered about #1100 :wink:

hhaoyan, 2019-12-02:
> Presumably just the following?
> 
> `sudo apt install bc`
> 
> Is that needed on 32 and 64 bit systems? I don't remember having to install it on Lubuntu, but it was a while ago.

I agree. But I don’t have a 32 bit Linux to test with.

I assume adding more dependencies won’t hurt? At least some people (like me) don’t have them.

JamesH65, 2019-12-02:
I'm just going to copy that line from #1100 in to the cross section as well.

JamesH65, 2019-12-02:
OK, done, and merged direct to master as such a small change.

lurch, 2019-12-02:
@JamesH65 Apologies for not explaining things very well - https://github.com/raspberrypi/documentation/blob/master/linux/kernel/building.md#build-sources already included the cross-compilation dependencies, so I've fixed things "properly" in https://github.com/raspberrypi/documentation/commit/9f4580ebee74fcfc13dbf891261995a278c6fae1

closed 2019-12-02 by lurch

JamesH65, 2019-12-02:
Ahhhhh, gotcha. Completely missed that line in the next section. Thanks.