raspberrypi/documentation #1357: set BOOT_UARt=1 but still not work

HuangShiqing, 2019-12-08:
I want to see the log from UART when my 3b+ module boot the file bootcode.bin, and I use the command `sed -i -e "s/BOOT_UART=0/BOOT_UART=1/" bootcode.bin` to enable the log print, but I get nothing from my ch340 device which is connecting to GPIO14 and GPIO15, I am sure my connection and setting with rpi is right because I can get some log when I login in the desktop  like below:
`Raspbian GNU/Linux 9 raspberrypi ttyS0`
`raspberrypi login:`

so I think my uart connection is right, but why I can't get log when boot the bootcode.bin? I have set the enable_uart=1 in config.txt. I  also try to download the latest firmware but still not work. could anybody tell me the reason?

lurch, 2019-12-08:
ping @ghollingworth @timg236

ghollingworth, 2019-12-08:
Have you checked the BOOT_UART is set properly?

strings /boot/bootcode.bin | grep BOOT

HuangShiqing, 2019-12-08:
> Have you checked the BOOT_UART is set properly?
> 
> strings /boot/bootcode.bin | grep BOOT

yes, I have followed the document command seriously, and after the sed command I use the strings command to check and get the result is 
`BOOT_UART=1`
that's the weirdest part about it.

JamesH65, 2019-12-09:
This is actually a repo concerning documentation, not for technical support. But I guess since we have started....

The right place to ask these questions is on the forum, or in the firmware repo.

HuangShiqing, 2019-12-11:
> This is actually a repo concerning documentation, not for technical support. But I guess since we have started....
> 
> The right place to ask these questions is on the forum, or in the firmware repo.

Thanks for your remind, I am going to ask the question in the forum

lurch, 2019-12-11:
> I am going to ask the question in the forum

Can this issue in the documentation repo be closed then?

HuangShiqing, 2019-12-12:
> > I am going to ask the question in the forum
> 
> Can this issue in the documentation repo be closed then?

sure, thanks again for every one

HuangShiqing, 2019-12-12:
This is actually a repo concerning documentation

closed 2019-12-12 by HuangShiqing