raspberrypi/documentation #1361: Special bootcode.bin-only boot mode

treble2019, 2019-12-16:
Please add 
https://github.com/raspberrypi/documentation/blob/master/hardware/raspberrypi/bootmodes/README.md#special-bootcodebin-only-boot-mode

to
https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/msd.md

PS: I am not 100 % sure that this mode works but may be adding it in documentation would lead more persons to test and report bugs!

lurch, 2019-12-16:
See #1360  :wink:

Paul-Ver, 2019-12-18:
@treble2019 thank you for your interest in making the documentation clearer and more available to all fellow pi-users.

As @lurch just pointed out; we're 'refactoring' the "USB Mass Storage boot" page at the moment.

In the coming version we have a reference/link to the "Special bootcode.bin boot mode".

We do not include the whole proces off the "special bootcode.bin only boot mode" in the "USB Mass Storage boot" page; to avoid having it documented on two locations.
But also; it's technically not "USB boot" as it still requires an SD card. But I agree on your point that pointing users of Pi's not supporting actual USB boot to this mode will be helpful.

You may, or,:
- Close this issue if this answer is satisfactory.
- Close this issue when #1360 has been applied.
- Let us know if (and why) you disagree on the above, it is open for discussion.
- Submit a pull request proposing changes to the documentation that you feel fit.

treble2019, 2019-12-21:
Thanks the the detailed reply. You all are helpful!

> We do not include the whole proces off the "special bootcode.bin only boot mode" in the "USB Mass Storage boot" page; to avoid having it documented on two locations.
But also; it's technically not "USB boot" as it still requires an SD card. But I agree on your point that pointing users of Pi's not supporting actual USB boot to this mode will be helpful.

I am not sure how many of RPi users are using it with large HDDs (may be you also do not) without micro SD card. Hence, the only thing I may request is that you could get some whatever 8TB or larger external HDDs (especially new year sales) and then test

1.  "USB Mass Storage boot"  YES/NO
2. special bootcode.bin only boot mode YES/NO

| Product | USB Mass Storage boot |  bootcode.bin only |
|---------|-----|-------|
| RPi Zero WH | - |  -| 
| RPi 3b | USB 2.0  | USB 2.0 | 

This would be very helpful. (Another 'tangential' issue is with your documentation is many of the HDDs available in the market are USB3 based (blue connector with type-A). After some *anecdotal* testing, it seems that RPi is happy with USB2.0 drives to externally boot but not USB3 drives (Say with a RPi3b or 3b+). This is NOT clear in your documentation nor the new pull request https://github.com/raspberrypi/documentation/pull/1360

Reference: https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/msd.md 

Since RPi is a great plug-n-play device (especially with NOOBS - the OS, I mean). It would be great to get any off-the-shelf HDD - write the OS. This would mean supporting USB 3 devices but may be this needs a new issue!

lurch, 2019-12-22:
AFAIK when you attach a USB3 device to a USB2 host, the host should just see the device as a regular USB2 device? :man_shrugging:  Backwards-compatibility and all that jazz :wink: 
Is the device externally-powered, or is it drawing power from the Pi? I think the USB3 specs allow devices to pull more power (from the host) than the USB2 specs do?

> Hence, the only thing I may request is that you could get some whatever 8TB or larger external HDDs (especially new year sales) and then test.

The problem with "compatibility lists" was addressed here - https://github.com/raspberrypi/documentation/pull/1360#issuecomment-565359171

@treble2019 Strictly speaking this repo is purely about the documentation. If you have issues with the firmware (i.e. not being able to boot from particular USB drives, even with `bootcode.bin` on an SD card) they should be opened at https://github.com/raspberrypi/firmware

treble2019, 2019-12-22:
> Backwards-compatibility and all that jazz 
.. hahaha!

> Is the device externally-powered, or is it drawing power from the Pi? I think the USB3 specs allow devices to pull more power (from the host) than the USB2 specs do?

Every HDD **even** 2.5 inches ones I always power with a external power - I mean the external boxes that come with 12V power. I also checked the external storage box with 1TB HDD that USB boots without any meddling. Only 4TB and larger have issues (see below). It is a reputed one I think (hope) (from Conrad Electronics).

> The problem with "compatibility lists" was addressed here

Agree with compatibility lists caveat. But I would like to know **at least** if the USB or bootcode.bin is applicable for USB2 or 3. At least some info. Not that user needs to buy every combination of device to get this done. You can write that we tested a typical USB-A type USB3 and it worked. 

Users would need some **base line** to compare. 

Rest assured, I really do **not** want you all to buy 20 different USB external devices from market.

>  If you have issues with the firmware (i.e. not being able to boot from particular USB drives, even with bootcode.bin on an SD card) they should be opened at https://github.com/raspberrypi/firmware

I did but still waiting... https://github.com/raspberrypi/firmware/issues/1304

closed 2019-12-22 by treble2019

lurch, 2019-12-23:
So based on your comments so far, we can't tell if it's the "USB3" that's causing the problem, or the "large drive" that's causing the problem? :man_shrugging: 

> Rest assured, I really do not want you all to buy 20 different USB external devices from market.

LOL, if it was only 20 it would probably be fairly easy, but there are many **many** more than that! :wink:

> I did but still waiting...

Good call, hopefully someone will be able to look into it in the new year :crossed_fingers: