raspberrypi/documentation #1372: Documentation refers to Pi 4 HDMI port 0 without specifying which that is.

HughR, 2020-01-05:
[HDMI Config](https://www.raspberrypi.org/documentation/configuration/hdmi-config.md) says that HDMI port 0 must be used for UltraHD at 60Hz.  But it isn't clear which of the two ports is 0.

I read the [schematic](https://www.raspberrypi.org/documentation/hardware/raspberrypi/schematics/rpi_SCH_4b_4p0_reduced.pdf) to find port 0 is labelled "j6".  This is silk-screened onto the circuit board.

I think that the clearest description is "the port closest to the USB C power connector".  This should be added to the HDMI Config page.

andrum99, 2020-01-06:
> I read the [schematic](https://www.raspberrypi.org/documentation/hardware/raspberrypi/schematics/rpi_SCH_4b_4p0_reduced.pdf) to find port 0 is labelled "j6". This is silk-screened onto the circuit board.

It's actually labelled HDMI0 on the silk screen. I've opened #1374 to address this.

JamesH65, 2020-01-06:
Fixed merged. Thanks all.

closed 2020-01-06 by JamesH65