raspberrypi/documentation #1387: SD Card size for full NOOBS

tracygardner, 2020-01-15:
Documentation says 8GB SD card is sufficient for NOOBS but with Buster this doesn't seem to give enough space to install full Raspbian in offline NOOBS. 

https://www.raspberrypi.org/documentation/installation/noobs.md
https://www.raspberrypi.org/documentation/installation/sd-cards.md

Full Raspbian with offline NOOBS is the recommended option in our beginner tutorial material so this could be misleading.

JamesH65, 2020-01-15:
Odd, we would expect things to easily fit in 8GB. Waiting for someone to get in so I can check with him.

XECDesign, 2020-01-15:
If that someone is me, he's off sick. Full Raspbian is quite heavy. I'd need to double check, but I believe OP is correct.

XECDesign, 2020-01-17:
Full NOOBS leaves 5104MB available. Raspbian Full asks for 6019MB. So no, an 8GB card is no longer enough. NOOBS lite with a Raspbian Full install should still be okay, but won't leave much room to play with.

JamesH65, 2020-01-17:
See here for fixup. If all OK will merge later.

https://github.com/raspberrypi/documentation/pull/1389

tracygardner, 2020-01-17:
Thanks guys. FYI The full install does work on 8GB card with NOOBS lite but I found that after an update the disk was close to 100% (close enough that df reported 100%), after an apt clean it was down to 89% but beginners wouldn't know to do that.

lurch, 2020-04-29:
Fixed by #1389 

@tracygardner there's some stuff about "apt clean" in https://github.com/raspberrypi/documentation/blob/master/raspbian/updating.md

closed 2020-04-29 by lurch