raspberrypi/documentation #1388: Cross-compiling Linux 5.4 requires an updated GCC

6by9, 2020-01-15:
https://github.com/raspberrypi/documentation/blob/master/linux/kernel/building.md
Cross compiling references
~/tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian/bin or ~/tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian-x64/bin

Those are gcc 4,8 and won't work for the 5.4 kernel.
It should now be ```~/tools/arm-bcm2708/arm-linux-gnueabihf/bin``` which symlinks to the latest toolchain (currently should be 4.9.3 in ~/tools/arm-bcm2708/arm-linux-gnueabihf/bin).

@pelwell please confirm I'm correct on these details.

pelwell, 2020-01-15:
That's correct. I note that it's an x64-hosted toolchain and that there is no equivalent x86 version, but I doubt that will be a problem.

6by9, 2020-01-15:
@JamesH65 Could you do the honours?

pelwell, 2020-01-15:
I'm sure James would take a PR...

JamesH65, 2020-01-15:
I would, or I can do it. I prefer the former.

JamesH65, 2020-02-25:
Fixed by https://github.com/raspberrypi/documentation/pull/1413

closed 2020-04-29 by lurch