raspberrypi/documentation #1392: H265 missing in codec list

ingotee, 2020-01-28:
The H265 codec introduced with the Raspberry Pi 4B is missing in the list of codecs. 

(But then vcgencmd does not report whether hardware H265 is enabled, anyway. I open another issue over there.)

JamesH65, 2020-01-28:
That because the H265 HW is not part of the videocore, it's an entirely new block on the chip. So the VC6 knows nothing about it. Therefor vcgencmd (which talks to the VC6) also knows nothing about it.

What page do you think needs to be updated?

ingotee, 2020-01-28:
Thanks, James!

I meant https://github.com/raspberrypi/documentation/blob/master/raspbian/applications/vcgencmd.md

I suggest you add what you just wrote to the paragraph about `codec_enabled` and suggest a different way to find out if H265 is available, as you do with RAM size.

JamesH65, 2020-01-28:
I've made a slight change to that page which will be updated shortly.  I'm not aware of any other mechanism for checking the H265 codec status, as far as I know it's permanently enabled.

ingotee, 2020-01-28:
Thanks. I thought about s.th. like checking for RPi 4B.

JamesH65, 2020-02-25:
Closing this issue as questions answered/issue resolved.

closed 2020-02-25 by JamesH65