raspberrypi/documentation #1393: Formatting SDXC cards for NOOBS

procount, 2020-01-31:
Since SD cards are getting cheaper all the time, the number of new users struggling to get NOOBS working with cards >32GB seems to be increasing.

Whilst there is documentation for how to format larger SD cards as FAT 32 (i.e. https://github.com/raspberrypi/documentation/blob/master/installation/sdxc_formatting.md), this page doesn't seems to be linked to from many of the other documentation page indexes, or pages that explain how to install NOOBS. I struggled to find it when trying to reference it for a forum post answer.

I suggest adding more links to the afore-mentioned page for each location where NOOBS is explained, to help try and avoid such forum posts and to give new users a better first time experience.

lurch, 2020-01-31:
A quick [search](https://github.com/raspberrypi/documentation/search?q=sdxc_formatting&type=Code) finds that it's currently linked to from two pages. Where else would you like the link added?

Even better, feel free to submit a PR :slightly_smiling_face:

procount, 2020-01-31:
I have to admit that I failed to spot it in `installation/noobs.md` when I was manually searching. My bad. But then maybe new users are failing too? Perhaps it just needs making more prominent? 🤷‍♂ 

But here is a page where it should be added->
https://projects.raspberrypi.org/en/projects/raspberry-pi-setting-up/3

lurch, 2020-01-31:
> But here is a page where it should be added->
https://projects.raspberrypi.org/en/projects/raspberry-pi-setting-up/3

That lives in a different repo (outside of this documentation repo), and unfortunately clicking on the "View project & license on GitHub" link at the bottom of that page gives a 404 :slightly_frowning_face: 
Any idea who the point of contact is for that @JamesH65 ?

MrSmartPi, 2020-02-16:
### RaspBerry Pi 4 Model B cannot boot from partition greater than 32 GB even when it is formatted in FAT32?

Hello. I am new to RasberryPi and started two days ago with my brandnew Raspberry Pi 4 Model B.
My SDXC-Card has 128 GB (SanDisk Extreme Pro UHS-1 A2) and I tried different methods to get it booting:
(Working on Windows 10 ...)
1. I tried MotionEyeOS which you install with Etcher. So you directly write an disk-image with all the needed partitions, and had no problems booting my RasPi.

2. After that I formatted my SDXC-Card with SD Formatter, as described here, to delete all data and than formatted it as FAT32 with the MiniTool Partition Wizard Free Edition.
Works fine so long...
Now copied Noobs from RaspBerry Pi Webpage:
https://downloads.raspberrypi.org/NOOBS_latest that was
Version:3.3.1
Release date:2020-02-14

Trying to boot nothing happens. Only the red LED got lid, no activity for green LED and also the huge partition does not get repartitoned.

I believe that it is not sufficient to format the SDXC-card > 32 GB with FAT32, more over you need to partiton it manually to a boot partion <64 GB or <=32 GB.

As I was not in the mood to partion manually I tried the 3rd way:

3. Again formatted the Card with SD Formatter back to 128 GB exFAT.
Wrote the RaspBian Buster image https://downloads.raspberrypi.org/raspbian_latest with Etcher on the SD-Card.
Now it has three partitions like with the MotionEyeOS image. The boot-partition is FAT32 and about  19 GB.
<- No problem booting my RaspBerry Pi

**Proposal:**
Maybe this should be validated and if true hint should be added to the documentation, to create a boot partition smaller than 64GB or smaller 32 GB.

JamesH65, 2020-11-19:
Not sure how to proceed, if at all, with this one. I have reported the broken link, but not sure what else needs doing.

lurch, 2020-11-19:
I splashed out on a 512GB microSD card a while ago, with the intention of testing all these different "big card compatibility" issues... which unfortunately I didn't get round to yet. I'll try to remember to have a play this evening...

procount, 2020-11-19:
This was why I created a set of images that could just be flashed to a large SD card to create a small FAT32 partition, so that NOOBS could simply be copied across afterwards.
( https://github.com/procount/fat32images )

But isn't all this moot now we have the RPI Imaging Utility? This can now flash an image of NOOBS, or it can even unpack the standard NOOBS.ZIP file. I use it all the time to write an initial PINN image to an SD card.

NOOBS (and PINN? :wink:)  should be added to the RPI imaging utility now that noobs has accidentally dropped off the new software download webpage.

andrum99, 2021-03-20:
For end-users preparing an SD card for use with a Raspberry Pi, NOOBs is no longer the recommended tool: it has been superceded by Raspberry Pi Imager. So this issue can be closed.

closed 2021-06-07 by aallan