raspberrypi/documentation #1396: Is there a GUI Guide?

closs290, 2020-02-11:
I don't find all of the icons, etc. to be intuitive. Is there a guide somewhere about what different things mean? For example, the red thermometer icon. Is this a fatal overheat error, or just generally letting me know it's hot? Or the red battery icon. I don't have a battery, so what is this supposed to mean?

lurch, 2020-02-12:
https://www.raspberrypi.org/documentation/configuration/warning-icons.md

JamesH65, 2020-02-12:
There are also the icons on the desktop taskbar which are currently not documented. I do wonder whether they need something somewhere. We don't provide any docs for using the desktop at all.

duckida, 2020-04-14:
WiFi = WiFi 
Bluetooth = Bluetooth
—percent = CPU Usage

duckida, 2020-04-14:
> There are also the icons on the desktop taskbar which are currently not documented. I do wonder whether they need something somewhere. We don't provide any docs for using the desktop at all.
> 
> 
> 
> edit: Raspberry Pi beginners guide is available online which has all the desktop stuff described.  https://magpi.raspberrypi.org/books/beginners-guide-3rd-ed
> 
> 

There  a 4th edition available

lurch, 2020-04-15:
@duckida Nope, there's only a 3rd edition of the Beginner's Guide, which @JamesH65 has linked to above.
I suspect you may be getting confused with the [User Guide](https://www.amazon.co.uk/Raspberry-Pi-User-Guide-4th/dp/1119264367/), which is an older (and now slightly outdated) book? :wink: 

@lucyhattersley @robthez While searching for "Raspberry Pi Beginner's Guide", [this page](https://magpi.raspberrypi.org/books/beginners-guide) also appears in the search results, which only links to v1 of the guide (with no mention of v3). Would it be possible to add a link from that page to the latest [v3](https://magpi.raspberrypi.org/books/beginners-guide-3rd-ed) page ?

JamesH65, 2020-11-19:
Closing as a Won't Fix.

closed 2020-11-19 by JamesH65