raspberrypi/documentation #1397: Rpi Wireless Access Point bridge setup documentation improvement

jpcornil-git, 2020-02-11:
Reference: https://github.com/raspberrypi/documentation/blob/master/configuration/wireless/access-point.md (bridge section)

- Remove the following section (this step is not required; it just creates a bridge using the command line but related configuration is not persistent nor a preliminary step for anything):

---
Add a new bridge, which in this case is called br0.

`sudo brctl addbr br0`

Connect the network ports. In this case, connect eth0 to the bridge br0.

`sudo brctl addif br0 eth0`

---

- Given the above you can also remove the bridge-utils installation and replace

---
`sudo apt install hostapd bridge-utils`

---
with:

---
`sudo apt install hostapd`

---

- In order to handle systems running/conflict with networking.service, I suggest to replace:

---
Finally, restart systemd-networkd:

`sudo systemctl restart systemd-networkd`

You can also use the brctl tool to verify that a bridge br0 has been created.

---
with:

---
Finally, (re)start systemd-networkd:

`sudo systemctl disable networking.service`
`sudo systemctl enable systemd-networkd`
`sudo systemctl restart systemd-networkd`

You can also use the bridge tool to verify that a bridge br0 has been created.

`bridge -d link`

---

JamesH65, 2020-02-12:
This would be much better submitted as a pull request.

jpcornil-git, 2020-02-12:
Indeed/agree, just created https://github.com/raspberrypi/documentation/pull/1398

lurch, 2020-02-12:
Probably better to wait until after #1333 has been merged?

jpcornil-git, 2020-02-12:
Sure, not a big deal to rework this on top of it

jpcornil-git, 2021-04-02:
Closing old issue (all fixed in https://www.raspberrypi.org/documentation/configuration/wireless/access-point-bridged.md)

closed 2021-04-02 by jpcornil-git