raspberrypi/documentation #1400: Wireless Access Point - Some explanation for IP address chosen required

edbird, 2020-02-13:
Path `documentation/configuration/wireless/access-point.md`

Line 44: The IP address chosen was `192.168.4.1/24`.

It is not immediately clear to me why this was chosen. My guess is that this Pi is on a network where the default address space/main local network address space is `192.168.1.X`, and there are an additional 2 networks `192.168.2.X`, `192.168.3.X`. Then the next sequential address is `192.168.4.X`, and `192.168.4.1` is chosen for the IP of the Access Point. (IP of Raspberry Pi's wireless interface.)

It is unlikely that a new user already has a network `192.168.4.X` in use - so for the vast majority just blindly following the instructions without thinking this is unlikely to be an issues, but some further explanation may be necessary here?

lurch, 2020-02-18:
I suggest we revisit this after #1333 has been merged?

aallan, 2021-06-07:
So #1333 got merged. Does it resolve this issue?

aallan, 2021-07-24:
Closed due to inactivity.

closed 2021-07-24 by aallan