raspberrypi/documentation #1407: Kernel build instructions don't advice to change LOCALVERSION

olifre, 2020-02-24:
I walked through the instructions (`building.md`) just now and wonder if the instructions should advice to set / hint on setting:
```
CONFIG_LOCALVERSION="-v7l-MY_CUSTOM_KERNEL"
```
If this is not done, existing modules from a distro installation in `/lib/modules/` might be overwritten, at least when a release tag is used as base for the self-compiled kernel. This may come as a surprise for users compiling their kernel for the first time. 

It might also be useful to hint on how to "go back" to the distro kernel (is it just reinstallation of `raspberrypi-kernel/`)?

lurch, 2020-02-24:
ping @pelwell and @popcornmix

pelwell, 2020-02-24:
It sounds like a reasonable suggestion - do you feel like submitting a Pull Request using your recent experience?

olifre, 2020-02-24:
@pelwell Yes, can do, just wanted to clarify first whether the change is considered reasonable. 
I'll prepare a PR then soon-ish. 

About the "how to go back to the distro kernel", do you have experience on this already?

pelwell, 2020-02-24:
I'm not the best person to comment since I either build my own or use the rpi-update builds, but `sudo apt install --reinstall raspberrypi-bootloader` should work.

olifre, 2020-02-24:
@pelwell Thanks. I'll then submit a PR for the first point (`LOCALVERSION`) in the next minutes, and as soon as I rollback to distro kernel, will try `sudo apt install --reinstall raspberrypi-bootloader`. If that also works well, I'll do another PR with that one.

JamesH65, 2020-02-25:
Fixed by #1408

olifre, 2020-03-20:
@pelwell and others:
I'm not finally in the reverting procedure. I found that the following steps are needed to get back to the previous state:
```
sudo apt install --reinstall raspberrypi-bootloader
sudo apt install --reinstall raspberrypi-kernel
rm /boot/bcm2835-rpi-*
rm /boot/bcm2837-rpi-3-b*
rm /boot/bcm2836-rpi-2-b.dtb
rm -rf /lib/modules/*-${LOCALVERSION}
```
This seems to catch everything (obviously, `LOCALVERSION` is also needed here). 

Dou you believe it is worthwhile documenting at the same place? As part of the same PR?

lurch, 2020-04-29:
#1408 has been merged, so closing this.

closed 2020-04-29 by lurch