raspberrypi/documentation #1409: Schematic missing for Raspberry Pi 4 Model B v1.1 and v1.2

Tyler-Ward, 2020-02-25:
The hardware schematics page for the Raspberry Pi 4 Model B only list schematics for the 4.0 version. 

The Revision page lists the avliable versions of the pi 4 as v1.1 and v1.2. I assume the v4.0 schematic should actually be the v1.1 schematic and the v1.2 doesent have its reduced schematics yet, as the v1.2 reportedly changes parts in this schematic it would be usefull to get the updated one.

JamesH65, 2020-02-25:
Schematics usually turn up eventually for each revision, but can take some weeks/months.

lurch, 2020-02-25:
@JamesH65 I assume that the Pi 4B '4.0' version referenced in the [schematics](https://github.com/raspberrypi/documentation/blob/master/hardware/raspberrypi/schematics/README.md) is actually independent from the Pi 4B '1.1' and '1.2' versions listed in the [revision codes](https://github.com/raspberrypi/documentation/blob/master/hardware/raspberrypi/revision-codes/README.md) ? I suspect that's the cause of the original issue here. It looks like for all the non-Pi4 schematics, the revisions on the different pages do match up?
ping @jimbojr

andrum99, 2020-05-20:
The revision on the Pi 4B schematic (4) looks like it is the PCB version, not the model revision (i.e. 1.1,1.2 etc). Since the PCB version is invisible to software, as I understand it, the schematic should probably specify the model revision(s) it apples to. (I believe the PCB version is printed on the top of the PCB but is hidden under the Ethrrnet jack).

JamesH65, 2020-05-20:
@jimbojr @Roger-Thornton Do we have any more recent schematics? Or a publishable reference to which PCB revision is in which board revision?

andrum99, 2020-05-20:
Just to clarify - by "the schematic should probably specify the model revision(s) it applies to" I was suggesting that instead of specifying the PCB revision, it should specify the model revision (v1.1 etc). (Not as well as). That would make it match how it is done on the other models of Pi.

SkullKill, 2020-07-06:
The Rev4.0 Schematic does not have the USB-C / CC1/ CC2, 5.1Kohm resistor fix. so there should be a newer version of the schematic which has not been published.

ericalbers, 2021-01-04:
As far as I can tell, they have not released the schematics for the Pi 4, they have some wiring diagram they call a schematic for the IO pins, but no actual schematics.  
I guess the schematics are proprietary/secret and not for release.

andrum99, 2021-01-05:
Or maybe the "Revision 4.0" label on the Pi 4B reduced schematic is simply version 4.0 of the reduced schematic for the Pi 4B, since it is sans USB-C resistor fix? 🤷‍♂️

aallan, 2021-06-07:
This will be resolved by #1911.

closed 2021-06-07 by aallan