raspberrypi/documentation #1420: TCP/IP documentation lacks IPv6 information

imp1sh, 2020-02-26:
The documentation found here `https://www.raspberrypi.org/documentation/configuration/tcpip/README.md`
lacks information about how IPv6 can be configured. 
I think it should cover:
- static IPv6
- SLAAC IPv6 and
- Statefull IPv6
dhcpcd should be able to hand all of those, right?

lurch, 2020-02-26:
Does #1318 cover what you're looking for?

imp1sh, 2020-02-26:
No.
The user needs information on how to give a raspbian host an IPv6 address. In default settings it's doing SLAAC but to do what if it wants a static IPv6 address or use dhcpv6 client?

JamesH65, 2020-02-26:
I have no idea how to do any of that, and don't have any time to acquire to required knowledge, so I would welcome a PR.

thomasschaeferm, 2020-03-06:
It is documented inside the `/etc/dhcpcd.conf` as comments.

aallan, 2021-06-07:
> Parts of that are included in my pull request. For dyndns I need a predictable/stable address.

Which PR is this?

thomasschaeferm, 2021-06-08:
#1318

closed 2021-06-22 by aallan