raspberrypi/documentation #1422: Noobs wont show all oses.

Jaccccc, 2020-03-01:
First, sorry if this is an issue in the wrong place, please tell me if i got this wrong, it wouldn't be the first time! So i recently got a raspberry pi 4, installed the latest noobs and installed raspbian. I now want to install windows IOT as well, but even when i connect to wifi, it wont show windows IOT anywhere?
Thanks!

lurch, 2020-03-02:
The NOOBS repo is over at https://github.com/raspberrypi/noobs and would be the correct place for reporting NOOBS issues.

However NOOBS on the Pi4 doesn't show Windows IOT, because AFAIK Microsoft haven't yet provided a version of Windows IOT that will run on the Pi4.
If you want more info about which OSes run on which models of Pi, you can take a look at https://downloads.raspberrypi.org/os_list_v3.json

JamesH65, 2020-03-02:
As @lurch says, there is no Windows 10 IOT for the Pi4. You would need to ask Microsoft why they have not done one, it's nothing  to do with us.

closed 2020-03-02 by JamesH65